secnotes/firmware-analysis-plus #2: 获取不到网卡应该如何解决

godzeo, 2020-11-10:
获取不到网卡应该如何解决
[image]

liyansong2018, 2020-11-10:
qemu能够模拟的外设有限，看截图，很有可能是这个固件的网卡，本身qemu就不支持，只能通过修改qemu源码，添加相应网卡来解决了。

godzeo, 2020-11-11:
那我可以测试一些你那个固件吗，我测试了三个，都是找不到那个网卡

godzeo, 2020-11-11:
我找了一下你那个固件。。还是不行
[image]

liyansong2018, 2020-11-12:
换个系统，跟你的环境有关系。我今天用Kali2020试了下，没有问题
[+] Firmware: wnap320_V3.7.11.4_firmware.tar
[+] Extracting the firmware...
[+] Image ID: 1
[+] Identifying architecture...
[+] Architecture: mipseb
[+] Building QEMU disk image...
[+] Setting up the network connection, please standby...
[+] Network interfaces: [('brtrunk', '192.168.0.100')]
[+] Using qemu-system-mips from /home/lys/Tools/firmware-analysis-plus-master/qemu-builds/2.5.0
[+] All set! Press ENTER to run the firmware...
[+] When running, press Ctrl + A X to terminate qemu

liyansong2018, 2020-11-12:
曾在Ubuntu18.04上也测试过，没有问题

godzeo, 2020-11-13:
好像是系统的问题，换了一个系统重装是可以的

closed 2021-02-28 by liyansong2018

Mingkko, 2021-04-23:
我在实验的过程中遇到了一个奇怪的问题。
系统是Ubuntu18.04。FAP v0.1
由于在模拟过程中按ctrl + x或a无法退出，于是使用kill -9 结束进程。第二次再次模拟的时候就会出现获取不到网卡的问题，运行reset.py也无法解决这个问题，但是重启一下服务器之后再次模拟又可以获取到网卡。
所以对这个问题还是挺疑惑的，不知道作者有没有什么解决的办法。