secnotes/firmware-analysis-plus #6: 您好，我在运行时发生了错误，我修改了密码和路径，但是还是错误帮忙看下呗

1604010326, 2021-01-30:
请问作者除了按博客里说的下载其他东西还需要下载吗，比如命令后面的zip

[+] Firmware: wnap320_V3.7.11.4_firmware.tar
[+] Extracting the firmware...
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/pexpect/spawnbase.py", line 150, in read_nonblocking
    s = os.read(self.child_fd, size)
OSError: [Errno 5] Input/output error

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/pexpect/expect.py", line 99, in expect_loop
    incoming = spawn.read_nonblocking(spawn.maxread, timeout)
  File "/usr/lib/python3/dist-packages/pexpect/pty_spawn.py", line 465, in read_nonblocking
    return super(spawn, self).read_nonblocking(size)
  File "/usr/lib/python3/dist-packages/pexpect/spawnbase.py", line 155, in read_nonblocking
    raise EOF('End Of File (EOF). Exception style platform.')
pexpect.exceptions.EOF: End Of File (EOF). Exception style platform.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "fat.py", line 168, in <module>
    main()
  File "fat.py", line 156, in main
    image_id = run_extractor(args.firm_path)
  File "fat.py", line 53, in run_extractor
    child.expect_exact("Tag: ")
  File "/usr/lib/python3/dist-packages/pexpect/spawnbase.py", line 390, in expect_exact
    return exp.expect_loop(timeout)
  File "/usr/lib/python3/dist-packages/pexpect/expect.py", line 105, in expect_loop
    return self.eof(e)
  File "/usr/lib/python3/dist-packages/pexpect/expect.py", line 50, in eof
    raise EOF(msg)
pexpect.exceptions.EOF: End Of File (EOF). Exception style platform.
<pexpect.pty_spawn.spawn object at 0x7f291750c4e0>
command: /usr/bin/python2
args: ['/usr/bin/python2', '/home/hadoop/firmware-analysis-plus/firmadyne/sources/extractor/extractor.py', '-np', '-nk', '/home/hadoop/firmware-analysis-plus/testcases/wnap320_V3.7.11.4_firmware.tar', '/home/hadoop/firmware-analysis-plus/firmadyne/images']
buffer (last 100 chars): b''
before (last 100 chars): b'ctor/extractor.py", line 18, in <module>\r\n    import binwalk\r\nImportError: No module named binwalk\r\n'
after: <class 'pexpect.exceptions.EOF'>
match: None
match_index: None
exitstatus: None
flag_eof: True
pid: 9782
child_fd: 5
closed: False
timeout: None
delimiter: <class 'pexpect.exceptions.EOF'>
logfile: None
logfile_read: None
logfile_send: None
maxread: 2000
ignorecase: False
searchwindowsize: None
delaybeforesend: 0.05
delayafterclose: 0.1
delayafterterminate: 0.1
searcher: searcher_string:
    0: "b'Tag: '"

1604010326, 2021-01-30:
您好，我再次安装了binwalk可以执行了但是
但是不显示network interface是啥，我不知到在网页中输入什么Ip网址来进入路由器网页管理界面

liyansong2018, 2021-01-30:
这说明没有正确获取路由器的IP地址，之前脚本是没有问题的，最近我也发现了这个问题，可能是由于某个库的更新，不知道你用的是什么系统

liyansong2018, 2021-01-30:
请参考 #2

1604010326, 2021-01-30:
> 这说明没有正确获取路由器的IP地址，之前脚本是没有问题的，最近我也发现了这个问题，可能是由于某个库的更新，不知道你用的是什么系统

ubuntu18.04,我用的一个tar固件文件，现在在尝试你这个zip正在下载

1604010326, 2021-01-30:
那别的固件文件不是路由器的也可以吗？如果分析别的固件，需要改什么地方吗

liyansong2018, 2021-01-30:
对于通用的固件，是不需要修改的，其他固件，各个厂商千差万别，所以定制比较复杂。最近我在研究`Firmadyne`的源码，有兴趣你也可以阅读，后续可能会出教程说明怎么定制。

liyansong2018, 2021-01-30:
> 您好，我再次安装了binwalk可以执行了但是
> 但是不显示network interface是啥，我不知到在网页中输入什么Ip网址来进入路由器网页管理界面

如果只是获取不到网卡，在已经启动路由器内核的情况下，可以输入 `ipconfig`，手动获取路由器的 ip 地址，然后修改 `scratch` 目录下的 `run.sh` 脚本中的 `qemu-system*` 命令，主要是针对 `net=tap`。手动在宿主机中创建网卡，将宿主机的网卡和 qemu 虚拟机的网卡连接起来。

1604010326, 2021-01-30:
> 对于通用的固件，是不需要修改的，其他固件，各个厂商千差万别，所以定制比较复杂。最近我在研究`Firmadyne`的源码，有兴趣你也可以阅读，后续可能会出教程说明怎么定制。

> > 您好，我再次安装了binwalk可以执行了但是
> > 但是不显示network interface是啥，我不知到在网页中输入什么Ip网址来进入路由器网页管理界面
> 
> 如果只是获取不到网卡，在已经启动路由器内核的情况下，可以输入 `ipconfig`，手动获取路由器的 ip 地址，然后修改 `scratch` 目录下的 `run.sh` 脚本中的 `qemu-system*` 命令，主要是针对 `net=tap`。手动在宿主机中创建网卡，将宿主机的网卡和 qemu 虚拟机的网卡连接起来。

好的，我看了下我的scratch目录下还有1,2,3三个文件夹，进入其中任意一个里面有run.sh，并没有qemu-system*相关命令啊~

liyansong2018, 2021-01-30:
`qemu` 启动系统的命令，其中有包括 `tap` 等。你是每个固件都获取不到网卡地址吗

1604010326, 2021-01-30:
> `qemu` 启动系统的命令，其中有包括 `tap` 等。你是每个固件都获取不到网卡地址吗

我换了个固件发现卡在下面这行不动了，很神奇，楼主可否建个群交流下哈哈，当然只是个建议，github发不了图片~
[+] Extracting the firmware...

1604010326, 2021-01-30:
> > `qemu` 启动系统的命令，其中有包括 `tap` 等。你是每个固件都获取不到网卡地址吗
> 
> 我换了个固件发现卡在下面这行不动了，很神奇，楼主可否建个群交流下哈哈，当然只是个建议，github发不了图片~
> [+] Extracting the firmware...

显示硬盘空间不足了hh我得阔下容

1604010326, 2021-01-31:
您好，我换了个固件试了下，报了如下错误。。。
[+] Extracting the firmware...
[+] Image ID: 2
[+] Identifying architecture...
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/pexpect/expect.py", line 99, in expect_loop
    incoming = spawn.read_nonblocking(spawn.maxread, timeout)
  File "/usr/lib/python3/dist-packages/pexpect/pty_spawn.py", line 462, in read_nonblocking
    raise TIMEOUT('Timeout exceeded.')
pexpect.exceptions.TIMEOUT: Timeout exceeded.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "fat.py", line 168, in <module>
    main()
  File "fat.py", line 161, in main
    arch = identify_arch(image_id)
  File "fat.py", line 80, in identify_arch
    child.expect_exact(":")
  File "/usr/lib/python3/dist-packages/pexpect/spawnbase.py", line 390, in expect_exact
    return exp.expect_loop(timeout)
  File "/usr/lib/python3/dist-packages/pexpect/expect.py", line 107, in expect_loop
    return self.timeout(e)
  File "/usr/lib/python3/dist-packages/pexpect/expect.py", line 70, in timeout
    raise TIMEOUT(msg)
pexpect.exceptions.TIMEOUT: Timeout exceeded.
<pexpect.pty_spawn.spawn object at 0x7f5d47f274e0>
command: /home/hadoop/firmware-analysis-plus/firmadyne/scripts/getArch.sh
args: ['/home/hadoop/firmware-analysis-plus/firmadyne/scripts/getArch.sh', '/home/hadoop/firmware-analysis-plus/firmadyne/images/2.tar.gz']
buffer (last 100 chars): b''
before (last 100 chars): b''
after: <class 'pexpect.exceptions.TIMEOUT'>
match: None
match_index: None
exitstatus: None
flag_eof: False
pid: 3297
child_fd: 5
closed: False
timeout: 30
delimiter: <class 'pexpect.exceptions.EOF'>
logfile: None
logfile_read: None
logfile_send: None
maxread: 2000
ignorecase: False
searchwindowsize: None
delaybeforesend: 0.05
delayafterclose: 0.1
delayafterterminate: 0.1
searcher: searcher_string:
    0: "b':'"

liyansong2018, 2021-01-31:
上午重新修改了一些脚本，等会上传，亲测 Kali 2020.04 可行。安装方法会更加简单，大幅缩减安装时间，下午我会重新更改说明文件，以及上传相应的安装说明。

1604010326, 2021-01-31:
> 上午重新修改了一些脚本，等会上传，亲测 Kali 2020.04 可行。安装方法会更加简单，大幅缩减安装时间，下午我会重新更改说明文件，以及上传相应的安装说明。

好的，感谢楼主，不知道ubuntu18.04可行不,下午可以也尝试一波

liyansong2018, 2021-01-31:
已提交最新代码，开源维护不易，请多多支持，当然不用buy me a coffee

1604010326, 2021-01-31:
> 已提交最新代码，开源维护不易，请多多支持，当然不用buy me a coffee

支持支持，If there is an entrance to buy coffee,  I am willing to buy it.
                                                                                                             ---thank you

1604010326, 2021-01-31:
github对国内网络太不友好了,龟速下载..
建议楼主复制到gitee一份~

1604010326, 2021-01-31:
楼主我执行这个 sudo ./deps.sh 命令下载巨慢，失败了好几次了，有什么好的解决方法吗

liyansong2018, 2021-02-01:
自家网络，正常运行，没有使用特殊渠道，下载畅通无阻。实在不行你可以试试科学上网

1604010326, 2021-02-01:
> 自家网络，正常运行，没有使用特殊渠道，下载畅通无阻。实在不行你可以试试科学上网

嗯嗯，只能这样了

1604010326, 2021-02-01:
楼主，这个binwalk在ubuntu18.04里还是安不上，一直报下面这个错
byte-compiling build/bdist.linux-x86_64/egg/ubireader/debug/__init__.py to __init__.cpython-36.pyc
  File "build/bdist.linux-x86_64/egg/ubireader/debug/__init__.py", line 27
    print obj.__name__, message
            ^
SyntaxError: Missing parentheses in call to 'print'. Did you mean print(obj.__name__, message)?

byte-compiling build/bdist.linux-x86_64/egg/ubireader/ubifs/misc.py to misc.cpython-36.pyc
  File "build/bdist.linux-x86_64/egg/ubireader/ubifs/misc.py", line 65
    except Exception, e:
                    ^
SyntaxError: invalid syntax

byte-compiling build/bdist.linux-x86_64/egg/ubireader/ubifs/output.py to output.cpython-36.pyc
  File "build/bdist.linux-x86_64/egg/ubireader/ubifs/output.py", line 44
    except Exception, e:
                    ^
SyntaxError: invalid syntax

byte-compiling build/bdist.linux-x86_64/egg/ubireader/ubifs/nodes/__init__.py to __init__.cpython-36.pyc
byte-compiling build/bdist.linux-x86_64/egg/ubireader/ubifs/display.py to display.cpython-36.pyc
byte-compiling build/bdist.linux-x86_64/egg/ubireader/ubifs/defines.py to defines.cpython-36.pyc
byte-compiling build/bdist.linux-x86_64/egg/ubireader/ubifs/walk.py to walk.cpython-36.pyc
  File "build/bdist.linux-x86_64/egg/ubireader/ubifs/walk.py", line 49
    except Exception, e:
                    ^
SyntaxError: invalid syntax

byte-compiling build/bdist.linux-x86_64/egg/ubireader/ubifs/__init__.py to __init__.cpython-36.pyc
  File "build/bdist.linux-x86_64/egg/ubireader/ubifs/__init__.py", line 59
    except Exception, e:
                    ^
SyntaxError: invalid syntax

byte-compiling build/bdist.linux-x86_64/egg/ubireader/utils.py to utils.cpython-36.pyc
byte-compiling build/bdist.linux-x86_64/egg/ubireader/settings.py to settings.cpython-36.pyc
byte-compiling build/bdist.linux-x86_64/egg/ubireader/ubi_io/__init__.py to __init__.cpython-36.pyc
  File "build/bdist.linux-x86_64/egg/ubireader/ubi_io/__init__.py", line 58
    except Exception, e:
                    ^
SyntaxError: invalid syntax

byte-compiling build/bdist.linux-x86_64/egg/ubireader/ubi/image/__init__.py to __init__.cpython-36.pyc
byte-compiling build/bdist.linux-x86_64/egg/ubireader/ubi/display.py to display.cpython-36.pyc
byte-compiling build/bdist.linux-x86_64/egg/ubireader/ubi/volume/__init__.py to __init__.cpython-36.pyc
byte-compiling build/bdist.linux-x86_64/egg/ubireader/ubi/defines.py to defines.cpython-36.pyc
byte-compiling build/bdist.linux-x86_64/egg/ubireader/ubi/headers/__init__.py to __init__.cpython-36.pyc
byte-compiling build/bdist.linux-x86_64/egg/ubireader/ubi/block/layout.py to layout.cpython-36.pyc
byte-compiling build/bdist.linux-x86_64/egg/ubireader/ubi/block/sort.py to sort.cpython-36.pyc
byte-compiling build/bdist.linux-x86_64/egg/ubireader/ubi/block/__init__.py to __init__.cpython-36.pyc
byte-compiling build/bdist.linux-x86_64/egg/ubireader/ubi/__init__.py to __init__.cpython-36.pyc
byte-compiling build/bdist.linux-x86_64/egg/ubireader/__init__.py to __init__.cpython-36.pyc
creating build/bdist.linux-x86_64/egg/EGG-INFO
installing scripts to build/bdist.linux-x86_64/egg/EGG-INFO/scripts
running install_scripts
running build_scripts
creating build/scripts-3.6
copying and adjusting scripts/ubireader_display_info -> build/scripts-3.6
copying and adjusting scripts/ubireader_extract_files -> build/scripts-3.6
copying and adjusting scripts/ubireader_extract_images -> build/scripts-3.6
copying and adjusting scripts/ubireader_utils_info -> build/scripts-3.6
changing mode of build/scripts-3.6/ubireader_display_info from 644 to 755
changing mode of build/scripts-3.6/ubireader_extract_files from 644 to 755
changing mode of build/scripts-3.6/ubireader_extract_images from 644 to 755
changing mode of build/scripts-3.6/ubireader_utils_info from 644 to 755
creating build/bdist.linux-x86_64/egg/EGG-INFO/scripts
copying build/scripts-3.6/ubireader_extract_images -> build/bdist.linux-x86_64/egg/EGG-INFO/scripts
copying build/scripts-3.6/ubireader_extract_files -> build/bdist.linux-x86_64/egg/EGG-INFO/scripts
copying build/scripts-3.6/ubireader_utils_info -> build/bdist.linux-x86_64/egg/EGG-INFO/scripts
copying build/scripts-3.6/ubireader_display_info -> build/bdist.linux-x86_64/egg/EGG-INFO/scripts
[... 22 lines cut ...]
Processing dependencies for ubi-reader==0.4
Finished processing dependencies for ubi-reader==0.4
正克隆到 'cramfs-tools'...
remote: Enumerating objects: 134, done.
remote: Total 134 (delta 0), reused 0 (delta 0), pack-reused 134
接收对象中: 100% (134/134), 65.27 KiB | 124.00 KiB/s, 完成.
处理 delta 中: 100% (80/80), 完成.
gcc -W -Wall -O2 -g -I.   mkcramfs.c  -lz -o mkcramfs
mkcramfs.c: In function ‘interpret_table_entry’:
mkcramfs.c:1263:5: warning: ignoring return value of ‘asprintf’, declared with attribute warn_unused_result [-Wunused-result]
     asprintf(&buf, "%s%lu", name, i);
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
gcc -W -Wall -O2 -g -I.   cramfsck.c  -lz -o cramfsck
cramfsck.c: In function ‘test_crc’:
cramfsck.c:229:4: warning: ignoring return value of ‘read’, declared with attribute warn_unused_result [-Wunused-result]
    read(fd, buf, super.size);
    ^~~~~~~~~~~~~~~~~~~~~~~~~
cramfsck.c: In function ‘romfs_read’:
cramfsck.c:299:3: warning: ignoring return value of ‘read’, declared with attribute warn_unused_result [-Wunused-result]
   read(fd, read_buffer, ROMBUFFERSIZE * 2);
   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cramfsck.c: In function ‘do_symlink’:
cramfsck.c:633:3: warning: ignoring return value of ‘asprintf’, declared with attribute warn_unused_result [-Wunused-result]
   asprintf(&str, "%s -> %s", path, outbuffer);
   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

liyansong2018, 2021-02-01:
白天还要打工，建议先使用 Kali 2020

1604010326, 2021-02-01:
> 白天还要打工，建议先使用 Kali 2020

好的，我觉得可能还是binwalk的事，一直就安的不大顺利

liyansong2018, 2021-02-28:
项目已更新，问题已解决

closed 2021-02-28 by liyansong2018