secnotes/firmware-analysis-plus #11: 运行错误，无法设置网络连接

MrBeanc, 2021-02-27:
环境：ubuntu18
mall0c@ubuntu:~/tools/firmware-analysis-plus$ sudo python3 fat.py -q ./2.5.0/ ~/router-test/dir645_FW_103.bin 
[sudo] mall0c 的密码： 

             
		______   _                ___                 
		|  ___| (_)              / _ \                
		| |_     _   _ __ ___   / /_\ \  _ __    ___  
		|  _|   | | | '_ ` _ \  |  _  | | '_ \  / __| ++
		| |     | | | | | | | | | | | | | | | | \__ \ 
		\_|     |_| |_| |_| |_| \_| |_/ |_| |_| |___/

                Welcome to the Firmware Analysis Plus - v2.0
            By lys - https://blog.csdn.net/song_lee | @liyansong
    
[+] Firmware: dir645_FW_103.bin
[+] Extracting the firmware...
[+] Image ID: 2
[+] Identifying architecture...
[+] Architecture: mipsel
[+] Building QEMU disk image...
[+] Setting up the network connection, please standby...
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/pexpect/spawnbase.py", line 150, in read_nonblocking
    s = os.read(self.child_fd, size)
OSError: [Errno 5] Input/output error

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/pexpect/expect.py", line 99, in expect_loop
    incoming = spawn.read_nonblocking(spawn.maxread, timeout)
  File "/usr/lib/python3/dist-packages/pexpect/pty_spawn.py", line 465, in read_nonblocking
    return super(spawn, self).read_nonblocking(size)
  File "/usr/lib/python3/dist-packages/pexpect/spawnbase.py", line 155, in read_nonblocking
    raise EOF('End Of File (EOF). Exception style platform.')
pexpect.exceptions.EOF: End Of File (EOF). Exception style platform.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "fat.py", line 168, in <module>
    main()
  File "fat.py", line 163, in main
    infer_network(arch, image_id, qemu_dir)
  File "fat.py", line 109, in infer_network
    child.expect_exact("Interfaces:", timeout=None)
  File "/usr/lib/python3/dist-packages/pexpect/spawnbase.py", line 390, in expect_exact
    return exp.expect_loop(timeout)
  File "/usr/lib/python3/dist-packages/pexpect/expect.py", line 105, in expect_loop
    return self.eof(e)
  File "/usr/lib/python3/dist-packages/pexpect/expect.py", line 50, in eof
    raise EOF(msg)
pexpect.exceptions.EOF: End Of File (EOF). Exception style platform.
<pexpect.pty_spawn.spawn object at 0x7f6cb2acc940>
command: /home/mall0c/tools/firmware-analysis-plus/firmadyne/scripts/inferNetwork.sh
args: ['/home/mall0c/tools/firmware-analysis-plus/firmadyne/scripts/inferNetwork.sh', '2', 'mipsel']
buffer (last 100 chars): b''
before (last 100 chars): b"icodeDecodeError: 'ascii' codec can't decode byte 0xc2 in position 5523: ordinal not in range(128)\r\n"
after: <class 'pexpect.exceptions.EOF'>
match: None
match_index: None
exitstatus: None
flag_eof: True
pid: 1662
child_fd: 5
closed: False
timeout: 30
delimiter: <class 'pexpect.exceptions.EOF'>
logfile: None
logfile_read: None
logfile_send: None
maxread: 2000
ignorecase: False
searchwindowsize: None
delaybeforesend: 0.05
delayafterclose: 0.1
delayafterterminate: 0.1
searcher: searcher_string:
    0: "b'Interfaces:'"
 我使用上游的项目时也卡在了Setting up the network connection, please standby...这一步，不知是什么原因，劳烦作者帮忙看一下

liyansong2018, 2021-02-28:
你这是解包产生的问题，首次运行FAP前，请确保以下两点：
1、**按照README中的说明，源码编译的方式安装binwalk，最好安装笔者改过的版本**。
2、`./setup.sh` （这一步你应该已经做过了）

liyansong2018, 2021-02-28:
另外，本项目的最新版只在 `Kali 2020` 版本上测试过最新版。由于 binwalk 依赖 python2 某些库的问题，firmadyne 不能兼容 Ubuntu 。（笔者发布的 FAP V1.0 解决了这个问题，可以运行在 Ubuntu 上，但是由于不再维护，不建议使用）

MrBeanc, 2021-02-28:
Kali 2020已经可以成功运行，但是我使用 ctrl + a 或者x似乎无法退出

liyansong2018, 2021-03-01:
是的，这是由于qemu提供的console无法正常退出造成的，可以使用替代方法
`ps -ef | grep qemu` + `kill -9 xxx`

MrBeanc, 2021-03-02:
好的，非常感谢

closed 2021-03-02 by MrBeanc

Mingkko, 2021-04-22:
遇到了相同的问题，我是Ubuntu 18.04，不知道是不是版本不兼容的问题，又不太想用FAP v1.0，等再尝试一下。

liyansong2018, 2021-04-22:
> 遇到了相同的问题，我是Ubuntu 18.04，不知道是不是版本不兼容的问题，又不太想用FAP v1.0，等再尝试一下。

建议在Kali下使用2.0