secnotes/firmware-analysis-plus #19: Image extraction failed

fishmisswater, 2021-07-23:
我在kali2021 02 上安装的，kali自带binwalk，默认管理员账户密码是kali:kali
按照说明步骤来，运行到这个一步：./fat.py -q ./2.5.0/ ./testcases/wnap320_V3.7.11.4_firmware.tar后报错了：
[+] Firmware: wnap320_V3.7.11.4_firmware.tar
[+] Extracting the firmware...
[!] Image extraction failed

liyansong2018, 2021-07-23:
由于项目中使用了binwalk提供的Python API，而Kali自带的binwalk提供的API有问题，所以解压失败。有两种解决方案
- 源码[编译binwalk](https://github.com/liyansong2018/firmware-analysis-plus#%E5%AE%89%E8%A3%85-binwalk)
- 使用firmware-analysis-plus的无binwalk模式，即使用`-b 0` [参数](https://github.com/liyansong2018/firmware-analysis-plus#%E7%BC%96%E8%AF%91-binwalk-%E5%A4%B1%E8%B4%A5%E6%80%8E%E4%B9%88%E5%8A%9E)

JsHuang, 2022-04-07:
Extract失败判断的逻辑是什么？我测试有的固件解压失败，会一直停留在`[+] Extracting the firmware...`很长时间 @liyansong2018

liyansong2018, 2022-04-07:
> Extract失败判断的逻辑是什么？我测试有的固件解压失败，会一直停留在`[+] Extracting the firmware...`很长时间 @liyansong2018

当前没有更加准确的方式，只是使用binwalk解压失败的输出信息来判断。可以直接使用binwalk解包，看看能否正确得到文件系统。

再使用工具进行仿真。参见主页：[编译 binwalk 失败怎么办？](https://github.com/liyansong2018/firmware-analysis-plus#%E7%BC%96%E8%AF%91-binwalk-%E5%A4%B1%E8%B4%A5%E6%80%8E%E4%B9%88%E5%8A%9E)

JsHuang, 2022-04-07:
> 当前没有更加准确的方式，只是使用binwalk解压失败的输出信息来判断。可以直接使用binwalk解包，看看能否正确得到文件系统。
>
> 再使用工具进行仿真。参见主页：[编译 binwalk [...]
在kali上编译binwalk成功，运行binwalk解压也可正确执行，但是使用Extractor却不能成功解压，可能原因是什么，怎么完全移除原有binwalk的影响

liyansong2018, 2022-04-07:
没有看到你具体的日志信息，不好判断。直接用 `-b` 参数，可以不使用 binwalk，跳过 Extractor

```shell
tar -czvf test.tar.gz *		# 一定要在固件文件系统的根目录下重新打包
./fat.py -q ./2.5.0/ -b 0  ./testcases/test.tar.gz
```

JsHuang, 2022-04-08:
@liyansong2018 错误信息如下
```
        >>>> Squashfs filesystem, little endian, version 4.0, compression:xz, size: 23724199 bytes, 2050 inodes, blocksize: 131072 bytes, created: 2021-11-30 02:55:53
        >>>> Extraction failed!

```

需要说明一下，你的binwalk源中，安装文件`deps.sh`, `python3-crypto`已经不存在这个库，最新的应该是`python3-cryptography`
此外binwalk自己下载的的sasquash目前直接编译会有问题,参见https://github.com/devttys0/sasquatch/issues/48
我是自行下载sasquash打补丁后安装的

liyansong2018, 2022-04-09:
@JsHuang 多谢提醒，抱歉，项目确实有一段时间没有维护了。是这样的，binwalk很多依赖没有跟上上游项目的进度，所以后来才提供了不编译binwalk也可以使用该工具的参数。

liyansong2018, 2022-04-09:
> Extract失败判断的逻辑是什么？我测试有的固件解压失败，会一直停留在`[+] Extracting the firmware...`很长时间 

通过在binwalk解压的目录中，寻找是否存在根文件系统来判断是否解压成功。如果文件系统存在如下文件夹，即符合要求。
```
UNIX_DIRS = ["bin", "etc", "dev", "home", "lib", "mnt", "opt", "root",
             "run", "sbin", "tmp", "usr", "var"]
```
因此这种判断方式存在一定的误报率。这也是基于工具特性设计的，因为如果固件中不包含文件系统，即使binwalk能够解压，firmware-analysis-plus 也无法仿真。

再次感谢你对binwalk安装时的一些上游项目已经更改的提醒！

commit bba531fd (2022-07-21): Close #19 #27 and #35
closed 2022-07-21 by liyansong2018