secnotes/firmware-analysis-plus #20: network interfaces为[]

buchiyexiao, 2021-07-28:
您好，我在使用fat.py进行模拟仿真的时候，会出现network interfaces为[]的现象
[+] Network interfaces: []
[+] All set! Press ENTER to run the firmware...
执行代码为sudo ./fat.py xx.bin
此前在对Dlink的某个版本仿真时没有出现这个问题，想问一下是系统配置有什么问题么

buchiyexiao, 2021-07-28:
（或者说，我们可以在什么位置自定义net来取代该情况

liyansong2018, 2021-07-28:
检查一下使用的 Linux 发行版是否符合项目主页上的[声明](https://github.com/liyansong2018/firmware-analysis-plus#firmware-analysis-plus)。你的问题看起来似乎跟 #13 是一样的。推荐使用 Kali，版本没有强制要求。这似乎是上游项目 `inferNetwork.sh` 在 Ubuntu 环境下存在的一个 bug，使用 Kali 应该能够解决该问题。

buchiyexiao, 2021-07-28:
（然后现在在kali上又出现了问题
Traceback (most recent call last):
  File "/home/bcyx/桌面/IOT/firmware-analysis-plus/./fat.py", line 182, in <module>
    main()
  File "/home/bcyx/桌面/IOT/firmware-analysis-plus/./fat.py", line 177, in main
    infer_network(arch, image_id, qemu_dir)
  File "/home/bcyx/桌面/IOT/firmware-analysis-plus/./fat.py", line 122, in infer_network
    child.expect_exact("Interfaces:", timeout=None)
  File "/usr/lib/python3/dist-packages/pexpect/spawnbase.py", line 421, in expect_exact
    return exp.expect_loop(timeout)
  File "/usr/lib/python3/dist-packages/pexpect/expect.py", line 179, in expect_loop
    return self.eof(e)
  File "/usr/lib/python3/dist-packages/pexpect/expect.py", line 122, in eof
    raise exc
pexpect.exceptions.EOF: End Of File (EOF). Exception style platform.
<pexpect.pty_spawn.spawn object at 0x7fa532c2c9d0>
command: /home/bcyx/桌面/IOT/firmware-analysis-plus/firmadyne/scripts/inferNetwork.sh
args: ['/home/bcyx/桌面/IOT/firmware-analysis-plus/firmadyne/scripts/inferNetwork.sh', '3', 'mipsel']
buffer (last 100 chars): b''
before (last 100 chars): b"Running firmware 3: terminating after 60 secs...\r\nqemu-system-mipsel: Invalid parameter 'vlan'\r\n"
after: <class 'pexpect.exceptions.EOF'>
match: None
match_index: None
exitstatus: None
flag_eof: True
pid: 12479
child_fd: 5
closed: False
timeout: 30
delimiter: <class 'pexpect.exceptions.EOF'>
logfile: None
logfile_read: None
logfile_send: None
maxread: 2000
ignorecase: False
searchwindowsize: None
delaybeforesend: 0.05
delayafterclose: 0.1
delayafterterminate: 0.1
searcher: searcher_string:
    0: b'Interfaces:'

buchiyexiao, 2021-07-28:
啊（不好意思，看到之前的解决办法了

closed 2021-07-28 by buchiyexiao

sdahiasdba, 2024-04-02:
anyone sovle this problem