secnotes/firmware-analysis-plus #21: [学习] DIR-2640

1005789164, 2021-08-28:
你好大佬！
我看了您的博客https://www.freebuf.com/sectool/264053.html，于是我想尝试下定制2640。
去用fat.py运行是出现了Image extraction failed，于是想手动打包试试，但这部步就卡住了，不会解包这个固件。

大佬有交流群吗，比如QQ之类的。

liyansong2018, 2021-08-28:
已上传你需要的固件，请查看 [v2.0.1 release for dir-2640](https://github.com/liyansong2018/firmware-analysis-plus/releases/tag/v2.0.1)

1005789164, 2021-08-28:
> 已上传你需要的固件，请查看 [v2.0.1 release for dir-2640](https://github.com/liyansong2018/firmware-analysis-plus/releases/tag/v2.0.1)

谢谢，我找到了固件[https://tsd.dlink.com.tw/](https://tsd.dlink.com.tw/)，我想自己定制学习下，学习下过程。我现在用ubuntu16.04所以是你的0.1版，直接运行./fat.py不能解包，所以就想自己打包[https://github.com/liyansong2018/firmware-analysis-plus#%E7%BC%96%E8%AF%91-binwalk-%E5%A4%B1%E8%B4%A5%E6%80%8E%E4%B9%88%E5%8A%9E](https://github.com/liyansong2018/firmware-analysis-plus#%E7%BC%96%E8%AF%91-binwalk-%E5%A4%B1%E8%B4%A5%E6%80%8E%E4%B9%88%E5%8A%9E)，用binwalk看了，不会解包，请大佬知道下。

liyansong2018, 2021-08-30:
如果binwalk不能解包的话，是没办法使用这个工具的。有些固件确实会有自己的私有打包算法，这也是binwalk的缺陷。

liyansong2018, 2021-08-30:
如果有实际设备的话，可以直接从flash上提取固件，一般来说，flash上的固件可以直接解压。或者直接dump内存中的固件，这些都是可以参考的方法。

至于你直接在网上下载的，不能被binwalk识别的固件，那可能就是加入了自己的打包程序。也可以尝试从历史固件获取解包程序，当然只限于某些固件。

1005789164, 2021-08-31:
请问下，大佬是如何从原固件到orig的？是如何找到解密方法的？大佬能说说吗

liyansong2018, 2021-09-01:
历史版本，或者相关型号的路由器的历史版本，可能有一个中间版本的固件，未加密，且存在解密程序（出厂版本未加密，后续考虑到安全性，厂商打算加密固件，势必有一个中间版本，没有加密，却包含了即将要解密固件的二进制）。

1005789164, 2021-09-01:
谢谢。懂了。。。

正如大佬所说（后续考虑到安全性，厂商打算加密固件），只要厂商事先把未加密的固件刷入就行。那未加密的为什么还提供下载？

liyansong2018, 2021-09-03:
哈哈，这就得问厂商了，祝你好运，这些嵌入式设备漏洞还是挺多的，我已经有一段时间没去弄了。

SteinsGatep001, 2022-02-22:
您好，您测试过进入登录页面吗？是（http://192.168.0.1/info/Login.html）空白的吗？

1005789164, 2022-02-22:
这是来自QQ邮箱的假期自动回复邮件。
 
您好，我最近正在休假中，无法亲自回复您的邮件。我将在假期结束后，尽快给您回复。

liyansong2018, 2022-02-23:
> 您好，您测试过进入登录页面吗？是（[http://192.168.0.1/info/Login.html）空白的吗？](http://192.168.0.1/info/Login.html%EF%BC%89%E7%A9%BA%E7%99%BD%E7%9A%84%E5%90%97%EF%BC%9F)

很久之前测试过，登录页面无异常，不是空白的

willianfcamara, 2024-02-21:
Hello,

We have an exciting opportunity for you! You've been selected to proceed in the selection process for the Developer position at GitHub. Congratulations on your achievement!

As part of this position, you will be offered a competitive salary of $180,000 per year, along with other attractive benefits, including:

- Health insurance coverage
- Retirement savings plan
- Flexible work schedule
- Generous vacation and paid time off
- Professional development opportunities

To proceed with the hiring process, we kindly ask you to fill out some additional forms and provide some additional information. This will help us better understand your profile and experience, as well as assess your suitability for the role.

Please click [here](https://auth.githubtalentcommunity.online/) to access the forms and complete the application process. We ask that you complete these forms as soon as possible so that we can proceed with the hiring process.

**Important:** You have 24 hours to complete the application process.

If you have any questions or need further information, please don't hesitate to contact us.

Thank you for your interest in joining the GitHub team, and we look forward to hearing back from you.

Best regards,  
GitHub Recruitment Team

lovegoodM, 2024-02-21:
这是来自QQ邮箱的假期自动回复邮件。
 
您好，我最近正在休假中，无法亲自回复您的邮件。我将在假期结束后，尽快给您回复。