secnotes/firmware-analysis-plus #22: 关于其他路由器的定制faq

guinsoo-code, 2021-09-12:
您好，关于faq对其他路由器如Archer_C50v5_US_0.9.1_0.2的定制化操作我仍没有头绪，打印日志里似乎并不是libnvram的问题，如果方便能提供Archer_C50v5_US_0.9.1_0.2或TL-WR802N(US)_V4_200的定制版faq吗？

liyansong2018, 2021-09-14:
最新版应该支持这两个固件，你试试直接git当前项目。

liyansong2018, 2021-09-14:
这个固件的定制非常简单，也是根据日志就可以定制，利用libnvram直接hook函数，参考我之前的 commit (7c2f6c8030b39341b6d5e8a2401a16bede1dc9a8)[https://github.com/liyansong2018/firmware-analysis-plus/commit/7c2f6c8030b39341b6d5e8a2401a16bede1dc9a8]

guinsoo-code, 2021-09-14:
好的，感谢你的回复，我确实看到libnvram里有相关注释和代码了，我现在怀疑是自己固件下载的有问题，我用自己下的Archer_C50v5仿真时，日志提示/sbin/mii-mgr的"ioctl error"，sys下某某目录不存在（dictionary  nonexistence）而无法创建文件（echo 512 > /sys/xxx/xxx/maxwaittime）以及访问的xml配置文件不存在，出现一堆read flash error -2等问题。还没有试TL-WR802N的效果。

liyansong2018, 2021-09-15:
我只在TL-WR802N上测试过，有时间我再适配下Archer_C50，你可以先尝试TL-WR802N，应该没有问题的。

guinsoo-code, 2021-09-15:
嗯，主要因为我对CVE-2021-29302感兴趣，想动手仿真复现下

liyansong2018, 2021-09-15:
这个漏洞上面就有个固件下载的链接，就是TL-WR802N，所以这个漏洞就是针对它的，当然系列路由器也是存在这个问题的

guinsoo-code, 2021-09-16:
我从https://www.tp-link.com/us/support/download/tl-wr802n/#Firmware页面下载了V4版提供的3个固件，使用的是最新的项目，结果都同Archer_C50v5一样未能成功仿真，且日志的报错信息也基本一样，是我下载的固件不对吗？
顺便我试过DIR2640的定制FAQ，并仿真成功，但admin@dlinkrouter的密码是什么呢？

guinsoo-code, 2021-09-16:
我从  https://www.tp-link.com/us/support/download/tl-wr802n/#Firmware  页面下载了V4版提供的3个固件，使用的是最新的项目，结果都同Archer_C50v5一样未能成功仿真，且日志的报错信息也基本一样，是我下载的固件不对吗？
顺便我试过DIR2640的定制FAQ，并仿真成功，但admin@dlinkrouter的密码是什么呢？

liyansong2018, 2021-09-16:
我刚刚又试了下，没有问题，我知道你说的报错信息了。
```shell
statSetsockopt
: Protocol not available
[ stat_getsockopt ] 219:  257 opt error
```
不用在意串口输出，此时你打开浏览器，输入 `192.168.0.1` 即可。这个路由器的固件没有提供串口shell，可以自己先解压路由器的文件系统，添加 busybox（telnetd），修改启动项，这样就可以提供 telnet 服务。

guinsoo-code, 2021-09-16:
懂了，一直看着这一串打印信息还以为是仿真失败了，再次谢谢你的答复了 ^ ^

da1sy, 2022-09-07:
你好，请问这里在使用你提供的固件进行模拟的时候，Interfaces结果为空是什么情况，我在多个固件的模拟过程中都有遇到过这个问题。

```
[da1sy@Pwn5ystem] ~/FAP (master) ⚡ 
❯ sudo ./fap.py -q ./qemu-builds/2.5.0 ~/TL_WR802N_debug.tar.gz 
[+] Firmware: TL_WR802N_debug.tar.gz
[+] Extracting the firmware...
[+] Image ID: 2
[+] Identifying architecture...
[+] Architecture: mipsel
[+] Building QEMU disk image...
[+] Setting up the network connection, please standby...
[+] [info] Running firmware 2: terminating after 60 secs...
qemu-system-mipsel: terminating on signal 2 from pid 6410
[+] [info] Inferring network...
[+] [info] Interfaces: []
[+] [info] Done!
[+] Using qemu-system-mipsel from /firmware-analysis-plus/qemu-builds/2.5.0
[+] All set! Press ENTER to run the firmware...
[+] When running, press Ctrl + A X to terminate qemu
[+] Command line: /firmware-analysis-plus/firmadyne/scratch/2/run.sh
....
....
statSetsockopt
: Protocol not available
[ stat_getsockopt ] 219:  257 opt error

statSetsockopt
: Protocol not available
[ stat_getsockopt ] 219:  257 opt error
```

ohyeah521, 2022-09-08:
弄个微信群吧，方便交流