secnotes/firmware-analysis-plus #25: gdb远程调试，貌似是提示架构不一致

huahai111, 2021-11-01:
![611f72c73f05699934b7df0b62520cc](https://user-images.githubusercontent.com/93123202/139677520-3a4fa6fb-4d79-450e-ba9d-16521c4877e1.png
[image]
)
连接上后，提示是图上的内容，没有办法调试，求师傅看看

huahai111, 2021-11-01:
[image]

huahai111, 2021-11-01:
gdb-multiarch busybox
GNU gdb (Debian 10.1-2) 10.1.90.20210103-git
Copyright (C) 2021 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from busybox...
(No debugging symbols found in busybox)
(gdb) target remote 192.168.0.1:4444
Remote debugging using 192.168.0.1:4444
warning: Selected architecture i386:x86-64 is not compatible with reported target architecture mips
warning: Architecture rejected target-supplied description
Reply contains invalid hex digit 59
(gdb) file busybox
warning: Selected architecture i386:x86-64 is not compatible with reported target architecture mips
Architecture of file not recognized.
(gdb) 

nc 192.168.0.1 23
��������
~ # ./gdbserver-7.12-mipsel-mips32rel2-v1 192.168.0.1:4444 ./bin/busybox
./gdbserver-7.12-mipsel-mips32rel2-v1 192.168.0.1:4444 ./bin/busybox
Process ./bin/busybox created; pid = 1683
Listening on port 4444
Remote debugging from host 192.168.0.2
Remote side has terminated connection.  GDBserver will reopen the connection.
Listening on port 4444
Remote debugging from host 192.168.0.2
Remote side has terminated connection.  GDBserver will reopen the connection.
Listening on port 4444
Remote debugging from host 192.168.0.2
Remote side has terminated connection.  GDBserver will reopen the connection.
Listening on port 4444

这是文字版的内容

liyansong2018, 2021-11-01:
请注意看你这一条日志
```
(gdb) target remote 192.168.0.1:4444
Remote debugging using 192.168.0.1:4444
warning: Selected architecture i386:x86-64 is not compatible with reported target architecture mips
```
这是由于你在使用`gdb-multiarch`的时候，后面直接跟了`busybox`，默认是Linux发行版，而不是固件中的`busybox`。所以宿主机即Linux上，也需要保留一份你需要调试的固件中的二进制。

`gdb-multiarch` 不能直接加载远程server上的符号表，所以需要在客户端使用如下命令
```
gdb-multiarch +目标二进制（而非宿主机上的二进制）
```

huahai111, 2021-11-02:
好的，感谢师傅

huahai111, 2021-11-02:
师傅，那这个以后我调试东西，必须要保留副本么。通过telnet传输到虚拟机或者，直接解压固件拷贝出来到虚拟机么

huahai111, 2021-11-02:
师傅，那这个以后我调试东西，必须要保留副本么。通过tftp传输到虚拟机或者，直接解压固件拷贝出来到虚拟机么

liyansong2018, 2021-11-02:
嗯，既然是调试，我相信你也需要将待调试的二进制通过IDA等软件反编译，这些都是常规操作

huahai111, 2021-11-02:
好的好的，谢谢师傅

liyansong2018, 2022-03-27:
关于调试，请参考 [固件远程登陆及二进制调试](https://github.com/liyansong2018/firmware-analysis-plus/wiki/%E5%9B%BA%E4%BB%B6%E8%BF%9C%E7%A8%8B%E7%99%BB%E9%99%86%E5%8F%8A%E4%BA%8C%E8%BF%9B%E5%88%B6%E8%B0%83%E8%AF%95)

closed 2022-03-27 by liyansong2018