secnotes/firmware-analysis-plus #27: 关于TL-WR802N固件仿真的问题

lsz9, 2021-12-04:
作者您好，很高兴发现这个这么优秀的作品。我正好尝试仿真TL-WR802N这个固件，我是直接使用firmadyme做仿真的，按照它的步骤，最后运行的时候一直报swRegRead: Operation not supported，但用您的工具仿真后仅报了几行，然后继续运行了，请问您在对这款固件做定制化开发的时候做了什么工作？非常感谢您

liyansong2018, 2021-12-05:
这个固件定制比较简单，修改几行代码即可，见 commit 7c2f6c8030b39341b6d5e8a2401a16bede1dc9a8。
不同固件定制的难度不一样，项目主页上有关于定制的一些文档。

lsz9, 2021-12-06:
作者您好，很高兴得到您的回复。我查看了您关于该款固件所做的定制化修复工作，其实就是在alias.c中新增了一个swClear然后重新编译成libnvram.so.mipel对吗？请问作为开发者，如何确定该款固件是需要这样一个函数来做劫持？请问思路是怎么样的？我更倾向于了解它的过程比起结论。

liyansong2018, 2021-12-06:
其实这篇文章 [开源固件仿真平台fap对嵌入式固件的模拟与定制](https://www.freebuf.com/sectool/264053.html) 已经给你答案了。

lsz9, 2021-12-07:
作者您好，我看了文章，大受启发。除此之外，正如上述问题所说，我直接尝试使用firmadyne进行仿真tp路由器的固件是仿真失败的，现在我知道问题了，应该是对一些nvram函数的操作没有劫持。我直接使用了您的2.0.2的fap版本中的firmadyne，发现在使用网络配置脚本配置时，居然配置失败了，但直接使用firmadyne时这一步是成功的。想请您指点一二，谢谢。
[image]

liyansong2018, 2021-12-07:
这似乎是上游项目脚本 inferNetwork.sh 的相关依赖在 Ubuntu 环境下存在的一个 bug，使用 Kali 应该能够解决该问题。

lsz9, 2021-12-13:
作者您好，使用您修复过的工具后，我已经可以成功使用firmadyne仿真TL-WR802N固件了。但是与其他路由器固件不同的是，我貌似没有在解析出的文件系统中找到与nvram有关的库或者字符串，是否tp路由器对于nvram有其具体的说法？您可以介绍下您当时针对这款固件采取了什么做法吗？谢谢

liyansong2018, 2021-12-18:
事实上，不同厂商可能在nvram库的基础上封装一层自己的函数，TP-Link可能就是这样，在标准nvram库的基础上又封装了若干函数。之前推荐的博文已经做了系统分析，定制是一个比较复杂的过程，需要对固件进行逆向，限于篇幅确实也没详细介绍各个厂商固件的定制。一般来说，仿真固件失败时，主要查看内核日志，根据日志定位到固件相应的代码，再根据代码，找到出错的地方。

closed 2022-07-21 by liyansong2018