secnotes/firmware-analysis-plus #28: kali2021.4模拟失败 qemu-system-mips: -net nic,vlan=0: Invalid parameter 'vlan'

Lantern-r, 2021-12-19:
````
──(kali㉿kali)-[~/Desktop/firmware-analysis-plus]
└─$ ./fat.py DIR.bin

             
                ______   _                ___                 
                |  ___| (_)              / _ \                
                | |_     _   _ __ ___   / /_\ \  _ __    ___  
                |  _|   | | | '_ ` _ \  |  _  | | '_ \  / __| ++
                | |     | | | | | | | | | | | | | | | | \__ \ 
                \_|     |_| |_| |_| |_| \_| |_/ |_| |_| |___/

                Welcome to the Firmware Analysis Plus - v2.1
 By lys - https://github.com/liyansong2018/firmware-analysis-plus | @liyansong
    
[+] Firmware: DIR.bin
[+] Extracting the firmware...
[+] Image ID: 1
[+] Identifying architecture...
[+] Architecture: mipseb
[+] Building QEMU disk image...
[+] Setting up the network connection, please standby...
/home/kali/Desktop/firmware-analysis-plus/firmadyne/scripts/inferNetwork.sh
---------
['1', 'mipseb']
---------
/home/kali/Desktop/firmware-analysis-plus/firmadyne
Traceback (most recent call last):
  File "/home/kali/Desktop/firmware-analysis-plus/./fat.py", line 184, in <module>
    main()
  File "/home/kali/Desktop/firmware-analysis-plus/./fat.py", line 179, in main
    infer_network(arch, image_id, qemu_dir)
  File "/home/kali/Desktop/firmware-analysis-plus/./fat.py", line 124, in infer_network
    child.expect_exact("Interfaces:", timeout=None)
  File "/usr/lib/python3/dist-packages/pexpect/spawnbase.py", line 421, in expect_exact
    return exp.expect_loop(timeout)
  File "/usr/lib/python3/dist-packages/pexpect/expect.py", line 179, in expect_loop
    return self.eof(e)
  File "/usr/lib/python3/dist-packages/pexpect/expect.py", line 122, in eof
    raise exc
pexpect.exceptions.EOF: End Of File (EOF). Exception style platform.
<pexpect.pty_spawn.spawn object at 0x7f90fb78d490>
command: /home/kali/Desktop/firmware-analysis-plus/firmadyne/scripts/inferNetwork.sh
args: ['/home/kali/Desktop/firmware-analysis-plus/firmadyne/scripts/inferNetwork.sh', '1', 'mipseb']
buffer (last 100 chars): b''
before (last 100 chars): b"mware 1: terminating after 60 secs...\r\nqemu-system-mips: -net nic,vlan=0: Invalid parameter 'vlan'\r\n"
after: <class 'pexpect.exceptions.EOF'>
match: None
match_index: None
exitstatus: None
flag_eof: True
pid: 10358
child_fd: 5
closed: False
timeout: 30
delimiter: <class 'pexpect.exceptions.EOF'>
logfile: None
logfile_read: None
logfile_send: None
maxread: 2000
ignorecase: False
searchwindowsize: None
delaybeforesend: 0.05
delayafterclose: 0.1
delayafterterminate: 0.1
searcher: searcher_string:
    0: b'Interfaces:'
````

Lantern-r, 2021-12-19:
qemu版本太高了...... 降版本就可以了

closed 2021-12-20 by Lantern-r

Ruoyyy, 2024-03-16:
> qemu版本太高了...... 降版本就可以了

如何降低版本