secnotes/firmware-analysis-plus #29: 登录的的账号和密码是多少？

Guo-yk, 2022-03-06:
博主你好，我已经仿真成功，浏览器也可以显示出登录页面，就是不知道username和password是多少？而且我在终端按回车键会出现让我login的密令，我的用户名是guoyk  密码是990405  当我输入的结束的时候会出现login incorrect，而且终端在一直运行，出现命令行，请问怎么让它不在一直出现命令行？

Guo-yk, 2022-03-06:
[image]
[image]
[image]

liyansong2018, 2022-03-06:
账号 root，密码 password

Guo-yk, 2022-03-06:
博主  我用了这个，但还是显示invalid username/password

liyansong2018, 2022-03-06:
试试
- Default Network name (SSID): NETGEAR_11ng
- Default Gateway IP: 192.168.0.100  
- Netgear login instructions
- Default username: admin  
- Netgear router default password list
- Default password: password 
- Default subnet mask: 255.255.255.0
- Default DNS server: 192.168.0.100 (Source: https://www.router-reset.com/info/Netgear/WNAP320#factory-default-settings)

Guo-yk, 2022-03-06:
可以了  谢谢博主

liyansong2018, 2022-03-06:
不客气~ 如果你想关闭路由器或者说命令行（路由器终端），请直接运行项目下的 `shutdown`

Guo-yk, 2022-03-07:
好的博主，如果我想运行其他支持的固件，我直接把它放在了testcases 目录下，比如DIR822A1_FW103WWb03.bin和TL_WR802N_debug.tar.gz然后运行，显示成功了，但是按了回车键就会报错，而且浏览器也访问不了，是不是要修改其他的配置，

liyansong2018, 2022-03-07:
尝试一下先运行 `reset.py` 清理一下空间，再运行 TL_WR802N_debug。理论上，不需要修改配置，也可以直接运行其他固件。

closed 2022-03-16 by liyansong2018