secnotes/firmware-analysis-plus #31: 关于 Tenda AC 15定制模拟的问题

huahai111, 2022-04-13:
参考了文章（https://www.freebuf.com/sectool/264053.html）中关于Tenda AC 15定制模拟的过程，在alias.c  config.h中添加了文章中所提到的信息：
[image]
[image]

然后，进行了编译，交叉编译工具是自己找的，编译成功了：
[image]

这里没有用文章中给的交叉编译工具，是因为提示：
firmware-analysis-plus/firmadyne/sources/arm-linux-musleabi/bin/../libexec/gcc/arm-linux-musleabi/5.3.0/cc1: error while loading shared libraries: libmpfr.so.4: cannot open shared object file: No such file or directory
make: *** [Makefile:14：console.o] 错误 1
[image]

然后开始模拟，但最终的模拟还是不行，似乎与没有编译修改前的报错是一致的（是对firmdyne的修改出了问题了么？）（报错日志不知道哪里是重要信息，所以都粘进去了），跪求师傅看一下是什么问题：
[    0.098105] [<c040f28c>] (kernel_init) from [<c0016da8>] (ret_from_fork+0x14/0x2c)
[    0.098222] ---[ end trace 4fc08b110b5ec178 ]---
[    0.098311] ------------[ cut here ]------------
[    0.098362] WARNING: CPU: 0 PID: 1 at /home/vagrant/firmadyne-kernel/kernel-v4.1/lib/kobject.c:240 kobject_add_internal+0x240/0x2ac()
[    0.098418] kobject_add_internal failed for gpio with -EEXIST, don't try to register things with the same name in the same directory.
[    0.098470] Modules linked in:
[    0.098526] CPU: 0 PID: 1 Comm: swapper Tainted: G        W       4.1.17+ #1
[    0.098562] Hardware name: Generic DT based system
[    0.098602] [<c001c99c>] (unwind_backtrace) from [<c0019d30>] (show_stack+0x10/0x14)
[    0.098650] [<c0019d30>] (show_stack) from [<c0024ab4>] (warn_slowpath_common+0x80/0xa8)
[    0.098701] [<c0024ab4>] (warn_slowpath_common) from [<c0024b08>] (warn_slowpath_fmt+0x2c/0x3c)
[    0.098755] [<c0024b08>] (warn_slowpath_fmt) from [<c018e834>] (kobject_add_internal+0x240/0x2ac)
[    0.098806] [<c018e834>] (kobject_add_internal) from [<c018e9a8>] (kset_register+0x1c/0x44)
[    0.098855] [<c018e9a8>] (kset_register) from [<c02090b4>] (__class_register+0xa8/0x198)
[    0.098904] [<c02090b4>] (__class_register) from [<c02091e4>] (__class_create+0x40/0x70)
[    0.098971] [<c02091e4>] (__class_create) from [<c01adf68>] (register_devfs_stubs+0x314/0xbb4)
[    0.099024] [<c01adf68>] (register_devfs_stubs) from [<c05d9b08>] (init_module+0x28/0xa4)
[    0.099092] [<c05d9b08>] (init_module) from [<c0009670>] (do_one_initcall+0x104/0x1b4)
[    0.099173] [<c0009670>] (do_one_initcall) from [<c05c7d08>] (kernel_init_freeable+0xf0/0x1b0)
[    0.099320] [<c05c7d08>] (kernel_init_freeable) from [<c040f28c>] (kernel_init+0x8/0xe4)
[    0.099489] [<c040f28c>] (kernel_init) from [<c0016da8>] (ret_from_fork+0x14/0x2c)
[    0.099661] ---[ end trace 4fc08b110b5ec179 ]---
[    0.099771] firmadyne: Cannot create device class: gpio!
[    0.100617] firmadyne: Cannot register character device: watchdog, 0xa, 0x82!
[    0.100815] firmadyne: Cannot register character device: wdt, 0xfd, 0x0!
[    0.144252] PCI host bridge /pcie@10000000 ranges:
[    0.144492]    IO 0x3eff0000..0x3effffff -> 0x00000000
[    0.144651]   MEM 0x10000000..0x3efeffff -> 0x10000000
[    0.144707]   MEM 0x8000000000..0xffffffffff -> 0x8000000000
[    0.145102] pci-host-generic 3f000000.pcie: resource collision: [mem 0x00000000-0xffffffff] conflicts with /pl011@9000000 [mem 0x09000000-0x09000fff]
[    0.145377] pci-host-generic: probe of 3f000000.pcie failed with error -16
[    0.151759] Non-volatile memory driver v1.3
[    0.163184] brd: module loaded
[    0.168105] loop: module loaded
[    0.173982]  vda: vda1
[    0.178413] 0.flash: Found 2 x16 devices at 0x0 in 32-bit bank. Manufacturer ID 0x000000 Chip ID 0x000000
[    0.178672] Intel/Sharp Extended Query Table at 0x0031
[    0.179167] Using buffer write method
[    0.179353] erase region 0: offset=0x0,size=0x80000,blocks=128
[    0.181552] 0.flash: Found 2 x16 devices at 0x0 in 32-bit bank. Manufacturer ID 0x000000 Chip ID 0x000000
[    0.181639] Intel/Sharp Extended Query Table at 0x0031
[    0.182037] Using buffer write method
[    0.182088] erase region 0: offset=0x0,size=0x80000,blocks=128
[    0.182151] Concatenating MTD devices:
[    0.182199] (0): "0.flash"
[    0.182236] (1): "0.flash"
[    0.182269] into device "0.flash"
[    0.187335] [nandsim] warning: read_byte: unexpected data output cycle, state is STATE_READY return 0x0
[    0.187426] [nandsim] warning: read_byte: unexpected data output cycle, state is STATE_READY return 0x0
[    0.187473] [nandsim] warning: read_byte: unexpected data output cycle, state is STATE_READY return 0x0
[    0.187519] [nandsim] warning: read_byte: unexpected data output cycle, state is STATE_READY return 0x0
[    0.187563] [nandsim] warning: read_byte: unexpected data output cycle, state is STATE_READY return 0x0
[    0.187624] [nandsim] warning: read_byte: unexpected data output cycle, state is STATE_READY return 0x0
[    0.187801] nand: device found, Manufacturer ID: 0x98, Chip ID: 0x39
[    0.187864] nand: Toshiba NAND 128MiB 1,8V 8-bit
[    0.187905] nand: 128 MiB, SLC, erase size: 16 KiB, page size: 512, OOB size: 16
[    0.188292] flash size: 128 MiB
[... 914 lines cut ...]
/dev/nvram: No such device
/dev/nvram: No such device
left wan[0]:0
left lan[0]:1
left lan[1]:2
left lan[2]:3
left lan[3]:4
vlan_array[0]: value 1  index_num 1 name  cpu_port 0index[0] 0  index[1] 0  index[2] 0  index[3] 0  index[4] 0  
vlan_array[1]: value 0  index_num 4 name  cpu_port 0index[0] 1  index[1] 2  index[2] 3  index[3] 4  index[4] 0  
vlan_array[2]: value -2  index_num 0 name  cpu_port 0index[0] 0  index[1] 0  index[2] 0  index[3] 0  index[4] 0  
vlan_array[3]: value -2  index_num 0 name  cpu_port 0index[0] 0  index[1] 0  index[2] 0  index[3] 0  index[4] 0  
vlan_array[4]: value -2  index_num 0 name  cpu_port 0index[0] 0  index[1] 0  index[2] 0  index[3] 0  index[4] 0  
nvram unset vlan1ports
/usr/sbin/wl: wl driver adapter not found
/usr/sbin/wl: wl driver adapter not found
The system is going down NOW!
Sent SIGTERM to all processes
Sent SIGKILL to all processes
Requesting system reboot
[    3.224440] cfg80211: Calling CRDA to update world regulatory domain
[    3.531646] firmadyne: sys_reboot[PID: 115 (init)]: magic1:fee1dead, magic2:28121969, cmd:1234567
[    3.532567] firmadyne: sys_reboot: removed CAP_SYS_BOOT, starting init...
[    6.372883] cfg80211: Calling CRDA to update world regulatory domain
[    9.522824] cfg80211: Calling CRDA to update world regulatory domain
QEMU: Terminated

liyansong2018, 2022-04-14:
交叉编译工具提示缺少库文件，可以进行下载。firmadyne 仿真Tenda AC15会有点问题。[ARM X](https://github.com/therealsaumil/emux)模拟器也是一个不错的选择。

huahai111, 2022-04-14:
好的，谢谢您的回复，我尝试一下ARM X模拟器

huahai111, 2022-04-14:
我尝试了eumx模拟运行，成功构建了docker执行了。在kali内访问localhost:20080成功访问到了tenda ac15的登录页面。
我还想问您一下，如果向docker中传入gdbserver，并与kali中的gdb建立连接进行调试，这需要怎么操作呢？我在eumx中好像没有找到相关的内容，烦请您指点😢

huahai111, 2022-04-14:
我找到了这些：
The run/ directory also contains a few commands that can be used from the host to interact with processes running within an EMUX emulated device.

    emuxhalt: Cleanly shut down the emulated device, and unmount all NFS mounts. Without a clean shutdown, there's always the risk of stale NFS handles.
    emuxps: Remotely enumerate processes running within EMUX.
    emuxmaps: Remotely dump the process memory layout of a process running within EMUX.
    emuxnetstat: Enumerate network sockets within EMUX.
    emuxkill: Remotely terminate a process running within EMUX.
    emuxgdb: Attach gdb to a process running within EMUX.
    monitor: Attach to the QEMU monitor.

emuxps, emuxmaps and emuxgdb are explained in detail in the [Debugging With EMUX](https://github.com/therealsaumil/emux/blob/master/docs/debugging-with-emux.html)[](https://github.com/therealsaumil/emux#contents-of-an-emulated-device) tutorial.
Contents of an emulated device

但是Debugging With EMUX指向的网页已经失效了

huahai111, 2022-04-14:
在docs找到了Debugging With EMUX。但gdbserver貌似封装再docker内部，想问您一下，可以向docker中映射目录或者导入文件么？还是只可以使用dockers本身封装的gdbserver呢？