secnotes/firmware-analysis-plus #32: 关于DIR860以上arm架构的固件无法正常模拟的问题

winmt, 2022-04-29:
我用的物理机是**Kali 2021.11**的版本，当我在模拟运行DIR-860L_v1.00的固件时，一切都**看似正常**地启动起来了：
[image]
这是一个**armel**架构的固件，而当我用nmap扫端口的时候，却发现**无法扫到任何开启的端口**，并且在Kali上也**无法ping通固件的网桥**：
[image]
之后，我又测试了DIR860以上版本的一些固件包，发现**mips架构都是可正常仿真模拟的，而arm架构都会出现上述的问题**。
不知道是否有其他师傅遇到过和我类似的问题，有没有什么比较好的解决办法呢？谢谢！

fxc233, 2022-05-01:
i meet the same problem

winmt, 2022-05-01:
我换成了官方所说的Kali 2020.04最适配版，依然存在这个问题。
能否麻烦作者师傅解答一下？谢谢！ @liyansong2018

fxc233, 2022-05-01:
The problem may be related to this error report

[image]

liyansong2018, 2022-05-02:
大家稍安勿躁，这个固件我之前并没有测试过，也不在主页的支持列表上。我来下载相应的固件看看这个问题，可能是上游firmadyne引入的。

winmt, 2022-05-02:
> 大家稍安勿躁，这个固件我之前并没有测试过，也不在主页的支持列表上。我来下载相应的固件看看这个问题，可能是上游firmadyne引入的。

多谢师傅，麻烦了。

liyansong2018, 2022-05-02:
已经修复了nvram部分问题，但是仍然不能ping通虚拟机网络。
```
iptables v1.4.7: can't initialize iptables table `nat': Table does not exist (do you need to insmod?)
Perhaps iptables or your kernel needs to be upgraded.
```
初步推断，是由arm版本的内核不支持某些特性导致的。
已经排除
- brctl 网桥配置问题
- route 路由问题

liyansong2018, 2022-05-04:
很遗憾，这个问题没有办法完全解决，这是firmadyne自身的bug。
- https://github.com/firmadyne/firmadyne/issues/61
- https://github.com/firmadyne/firmadyne/issues/136

winmt, 2022-05-05:
> 很遗憾，这个问题没有办法完全解决，这是firmadyne自身的bug。
> 
> * [Can't start web server firmadyne/firmadyne#61](https://github.com/firmadyne/firmadyne/issues/61)
> * [2 Bridge IPs but none of them is accessible firmadyne/firmadyne#136](https://github.com/firmadyne/firmadyne/issues/136)

收到，谢谢。
另外，请问这是D-Link系列arm架构路由firmadyne不支持，还是大多数arm架构的路由firmadyne都不支持呢？

liyansong2018, 2022-05-05:
应该是部分路由器不支持。这个问题后续可以进一步更进，看看能否有更好的解决方案。

目前来看，固件中的服务都能够正常拉起，就是宿主机和虚拟机之间的网络不通。

winmt, 2022-05-05:
> 应该是部分路由器不支持。这个问题后续可以进一步跟进，看看能否有更好的解决方案。
> 
> 目前来看，固件中的服务都能够正常拉起，就是宿主机和虚拟机之间的网络不通。

好的，谢谢您的回复。

liyansong2018, 2022-05-07:
@winmt @fxc233 已修复该问题，请运行 `git fetch`  同步项目最近更新。

liyansong2018, 2022-05-07:
@fxc233 This bug has been fixed and you can try the latest version.

closed 2022-05-07 by liyansong2018

winmt, 2022-05-07:
> @winmt @fxc233 已修复该问题，请运行 `git fetch` 同步项目最近更新。

已测试，可正常启动，并能成功打通一些CVE，感谢您的帮助！

fxc233, 2022-05-07:
@liyansong2018  Thanks for the quick fix and release!

commit 4b28751a (2022-05-07): Fix #32 about armel network error