secnotes/firmware-analysis-plus #33: 关于复现CVE-2019-17621 Dlink-859时遇到的问题

Dyingchen, 2022-05-21:
根据作者提供的链接教程进行的模拟，由于是初学搞不清楚发生了什么问题 
日志如下：
[+] Firmware: DIR822A1_FW103WWb03.bin
[+] Extracting the firmware...
[+] Image ID: 2
[+] Identifying architecture...
[+] Architecture: mipseb
[+] Building QEMU disk image...
[+] Setting up the network connection, please standby...
Traceback (most recent call last):
  File "./fap.py", line 185, in <module>
    main()
  File "./fap.py", line 180, in main
    infer_network(arch, image_id, args.time, qemu_dir)
  File "./fap.py", line 124, in infer_network
    child.expect_exact("Interfaces:", timeout=None)
  File "/usr/lib/python3/dist-packages/pexpect/spawnbase.py", line 421, in expect_exact
    return exp.expect_loop(timeout)
  File "/usr/lib/python3/dist-packages/pexpect/expect.py", line 179, in expect_loop
    return self.eof(e)
  File "/usr/lib/python3/dist-packages/pexpect/expect.py", line 122, in eof
    raise exc
pexpect.exceptions.EOF: End Of File (EOF). Exception style platform.
<pexpect.pty_spawn.spawn object at 0x7f0f48aa33a0>
command: /dycfiles/iot/firmware-analysis-plus/firmadyne/scripts/inferNetwork.sh
args: ['/dycfiles/iot/firmware-analysis-plus/firmadyne/scripts/inferNetwork.sh', '2', 'mipseb', '60']
buffer (last 100 chars): b''
before (last 100 chars): b" '/dycfiles/iot/firmware-analysis-plus/firmadyne/scratch//2//image.raw': No such file or directory\r\n"
after: <class 'pexpect.exceptions.EOF'>
match: None
match_index: None
exitstatus: None
flag_eof: True
pid: 13554
child_fd: 5
closed: False
timeout: 30
delimiter: <class 'pexpect.exceptions.EOF'>
logfile: None
logfile_read: None
logfile_send: None
maxread: 2000
ignorecase: False
searchwindowsize: None
delaybeforesend: 0.05
delayafterclose: 0.1
delayafterterminate: 0.1
searcher: searcher_string:
    0: b'Interfaces:'

liyansong2018, 2022-05-22:
删除缓存，再尝试一下。`./reset.py`。这个固件的模拟是支持的。

Dyingchen, 2022-05-22:
> 删除缓存，再尝试一下。 `./reset.py`。这个固件的模拟是支持的。

没有效果，之前就尝试过了 依旧是这样的

Dyingchen, 2022-05-22:
> 删除缓存，再尝试一下。 `./reset.py`。这个固件的模拟是支持的。

已解决，应该是qemu的依赖问题，重装了一遍之后就正常了

closed 2022-05-22 by Dyingchen