secnotes/firmware-analysis-plus #34: 关于固件加载失败问题

flamingo1616, 2022-05-24:
您好，
1. 我昨天在kali运行项目时一直卡在网络连接，最后更新了系统的一些文件，现在又卡住了。

[image]

希望您能帮我看一看

liyansong2018, 2022-05-24:
我尝试使用 `-b0` 以及 `-b1` 测试你的固件，仿真均没有问题。
```shell
$ ./fap.py -q ./2.5.0/ ./testcases/DIR823GA1_FW102B03.bin                                                            2 ⚙

             
            ______   _                ___                 
            |  ___| (_)              / _ \                
            | |_     _   _ __ ___   / /_\ \  _ __    ___  
            |  _|   | | | '_ ` _ \  |  _  | | '_ \  / __| ++
            | |     | | | | | | | | | | | | | | | | \__ \ 
            \_|     |_| |_| |_| |_| \_| |_/ |_| |_| |___/

            Welcome to the Firmware Analysis Plus - v2.2
 By lys - https://github.com/liyansong2018/firmware-analysis-plus
    
[+] Firmware: DIR823GA1_FW102B03.bin
[+] Extracting the firmware...
[+] Image ID: 1
[+] Identifying architecture...
[+] Architecture: mipsel
[+] Building QEMU disk image...
[+] Setting up the network connection, please standby...
[+] Network interfaces: [('br0', '192.168.0.1'), ('br1', '192.168.100.1')]
[+] Using qemu-system-mipsel from /home/lys/Tools/firmware-analysis-plus/qemu-builds/2.5.0
[+] All set! Press ENTER to run the firmware...
[+] When running, press Ctrl + A X to terminate qemu
[+] Command line: /home/lys/Tools/firmware-analysis-plus/firmadyne/scratch/1/run.sh

```
没有更多的日志支撑，你的问题跟环境有关系，确认下单独使用binwalk是否能够成功解压。

flamingo1616, 2022-05-24:
您好，这边我使用binwalk 解压你提供的测试固件，是可以解压到文件的
[image]

[image]

liyansong2018, 2022-05-24:
在**squashfs-root目录**下，重新压缩固件，使用 `./fap.py -q ./2.5.0/ -b false ./testcases/squashfs.tar.gz` ，再次测试。

liyansong2018, 2022-05-24:
或者切换普通身份（非root用户）再次尝试。

liyansong2018, 2022-05-24:
抱歉，这可能是引入的一个bug，正在调试中

flamingo1616, 2022-05-24:
没事，谢谢你的帮助。

flamingo1616, 2022-05-24:
好的我现在再测试一下，太打扰你了。

flamingo1616, 2022-05-24:
哈哈哈，我这边怕是寄了。

[image]

┌──(kali㉿kali)-[~/firmware/firmware-analysis-plus]
└─$ ./setup.sh                                             
[sudo] password for kali: 
Hit:1 http://mirrors.ustc.edu.cn/kali kali-rolling InRelease
Hit:2 http://mirrors.aliyun.com/kali kali-rolling InRelease
Hit:3 https://dl.google.com/linux/chrome/deb stable InRelease
Reading package lists... Done                    
Building dependency tree... Done
Reading state information... Done
261 packages can be upgraded. Run 'apt list --upgradable' to see them.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
busybox-static is already the newest version (1:1.30.1-7+b3).
fakeroot is already the newest version (1.28-1).
kpartx is already the newest version (0.8.8-1).
python3-pexpect is already the newest version (4.8.0-2).
python3-pip is already the newest version (22.1+dfsg-1).
snmp is already the newest version (5.9.1+dfsg-1+b1).
uml-utilities is already the newest version (20070815.4-1).
unzip is already the newest version (6.0-26).
util-linux is already the newest version (2.38-4).
vlan is already the newest version (2.0.5).
0 upgraded, 0 newly installed, 0 to remove and 261 not upgraded.
Looking in indexes: https://pypi.mirrors.ustc.edu.cn/simple/
Requirement already satisfied: python-magic in /usr/lib/python3/dist-packages (0.4.25)
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv             
=====================================================                                                                
Firmware Analysis Plus installed successfully!
Before running fap.py for the first time,
please edit fap.config and provide your sudo password
=====================================================

┌──(kali㉿kali)-[~/firmware/firmware-analysis-plus]
└─$ ./fap.py -q ./2.5.0/ ./testcases/iot_dir880l_110b01.bin

             
            ______   _                ___                 
            |  ___| (_)              / _ \                
            | |_     _   _ __ ___   / /_\ \  _ __    ___  
            |  _|   | | | '_ ` _ \  |  _  | | '_ \  / __| ++
            | |     | | | | | | | | | | | | | | | | \__ \ 
            \_|     |_| |_| |_| |_| \_| |_/ |_| |_| |___/

            Welcome to the Firmware Analysis Plus - v2.2
 By lys - https://github.com/liyansong2018/firmware-analysis-plus
    
[+] Firmware: iot_dir880l_110b01.bin
[+] Extracting the firmware...
[+] Image ID: 1
[+] Identifying architecture...
[+] Architecture: armel
[+] Building QEMU disk image...
Traceback (most recent call last):
  File "/home/kali/firmware/firmware-analysis-plus/./fap.py", line 185, in <module>
    main()
  File "/home/kali/firmware/firmware-analysis-plus/./fap.py", line 179, in main
    make_image(arch, image_id)
  File "/home/kali/firmware/firmware-analysis-plus/./fap.py", line 108, in make_image
    child.expect_exact(pexpect.EOF)
  File "/usr/lib/python3/dist-packages/pexpect/spawnbase.py", line 421, in expect_exact
    return exp.expect_loop(timeout)
  File "/usr/lib/python3/dist-packages/pexpect/expect.py", line 181, in expect_loop
    return self.timeout(e)
  File "/usr/lib/python3/dist-packages/pexpect/expect.py", line 144, in timeout
    raise exc
pexpect.exceptions.TIMEOUT: Timeout exceeded.
<pexpect.pty_spawn.spawn object at 0x7fdbfe0efcd0>
command: /usr/bin/sudo
args: ['/usr/bin/sudo', '--', '/home/kali/firmware/firmware-analysis-plus/firmadyne/scripts/makeImage.sh', '1', 'armel']
buffer (last 100 chars): b'[sudo] password for kali: \r\nSorry, try again.\r\n[sudo] password for kali: '
before (last 100 chars): b'[sudo] password for kali: \r\nSorry, try again.\r\n[sudo] password for kali: '
after: <class 'pexpect.exceptions.TIMEOUT'>
match: None
match_index: None
exitstatus: None
flag_eof: False
pid: 5736
child_fd: 5
closed: False
timeout: 30
delimiter: <class 'pexpect.exceptions.EOF'>
logfile: None
logfile_read: None
logfile_send: None
maxread: 2000
ignorecase: False
searchwindowsize: None
delaybeforesend: 0.05
delayafterclose: 0.1
delayafterterminate: 0.1
searcher: searcher_string:
    0: EOF

KongLynn, 2022-06-19:
@flamingo1616 您好，请问这个问题你是怎么解决的呢？我也遇到了同样的问题。

flamingo1616, 2022-06-20:
> @flamingo1616 您好，请问这个问题你是怎么解决的呢？我也遇到了同样的问题。

我暂时没有使用FAP了，这个我使用的时候兼容性还是有点问题。
你仿真固件的话，可以看看Attify-OS这个系统（IOT分析），里面有FAT环境和其他工具。

liyansong2018, 2022-06-20:
> @flamingo1616 您好，请问这个问题你是怎么解决的呢？我也遇到了同样的问题。

你用的也是Fap自带的测试固件么？

flamingo1616, 2022-06-20:
是的，貌似并不能成功的运行

liyansong2018, 2022-06-20:
我在Windows + Vmware（Kali Linux 2022）上测试过Fap，没有遇到你们的问题。你的运行环境是什么？

KongLynn, 2022-06-21:
> 我在Windows + Vmware（Kali Linux 2022）上测试过Fap，没有遇到你们的问题。你的运行环境是什么？@[KongLynn](https://github.com/KongLynn)

我的运行环境也是Windows + Vmware（Kali Linux 2022.2），报错如下：
使用binwalk：
[image]
无binwalk：
[image]

liyansong2018, 2022-06-21:
@KongLynn @flamingo1616 
非常抱歉，这是由 e502470d36bd4c8cd1658080f273e097714b5af0 引入的一个bug，已修复。如果问题还没解决，可以重新打开这个issue。如果有其他问题，欢迎提交新 issue。

liyansong2018, 2022-06-22:
该 Bug 由缺少依赖 qemu-utils 引起的。此工具用于创建 qemu 镜像文件，是必不可少的工具。因此该 issue 的问题是某次更新 e502470d36bd4c8cd1658080f273e097714b5af0 删除该依赖导致的，实际上是创建 qemu 镜像失败了，已修复 522246185747438f0db482f81c9ae53331137f4b。再次感谢大家的反馈！