secnotes/firmware-analysis-plus #35: 关于TOTOLink仿真失败的问题

kal1x, 2022-06-05:
- [固件地址](http://www.totolink.cn/data/upload/20210717/bc2b1d2206c4a7b1753c3d870cb15bff.zip)

- kali版本为最新：2022.1
- 将默认超时时间更改为了90

```
┌──(root㉿kali)-[/home/kali/Desktop/firmware-analysis-plus]
└─# ./fap.py ./testcases/TOTOLINK_C8540R-1C_A7100RU_IP04365_MT7621AMT7615Ex2_SPI_16M128M_V7.4cu.2313_B20191024_ALL.web 

             
            ______   _                ___                 
            |  ___| (_)              / _ \                
            | |_     _   _ __ ___   / /_\ \  _ __    ___  
            |  _|   | | | '_ ` _ \  |  _  | | '_ \  / __| ++
            | |     | | | | | | | | | | | | | | | | \__ \ 
            \_|     |_| |_| |_| |_| \_| |_/ |_| |_| |___/

            Welcome to the Firmware Analysis Plus - v2.2
 By lys - https://github.com/liyansong2018/firmware-analysis-plus
    
[+] Firmware: TOTOLINK_C8540R-1C_A7100RU_IP04365_MT7621AMT7615Ex2_SPI_16M128M_V7.4cu.2313_B20191024_ALL.web
[+] Extracting the firmware...
[+] Image ID: 1
[+] Identifying architecture...
[+] Architecture: mipsel
[+] Building QEMU disk image...
[+] Setting up the network connection, please standby...
/home/kali/Desktop/firmware-analysis-plus/firmadyne/scripts/inferNetwork.sh
---------
['1', 'mipsel', '90']
---------
/home/kali/Desktop/firmware-analysis-plus/firmadyne
Traceback (most recent call last):
  File "/home/kali/Desktop/firmware-analysis-plus/./fap.py", line 185, in <module>
    main()
  File "/home/kali/Desktop/firmware-analysis-plus/./fap.py", line 180, in main
    infer_network(arch, image_id, args.time, qemu_dir)
  File "/home/kali/Desktop/firmware-analysis-plus/./fap.py", line 124, in infer_network
    child.expect_exact("Interfaces:", timeout=None)
  File "/usr/lib/python3/dist-packages/pexpect/spawnbase.py", line 421, in expect_exact
    return exp.expect_loop(timeout)
  File "/usr/lib/python3/dist-packages/pexpect/expect.py", line 179, in expect_loop
    return self.eof(e)
  File "/usr/lib/python3/dist-packages/pexpect/expect.py", line 122, in eof
    raise exc
pexpect.exceptions.EOF: End Of File (EOF). Exception style platform.
<pexpect.pty_spawn.spawn object at 0x7f322d259d50>
command: /home/kali/Desktop/firmware-analysis-plus/firmadyne/scripts/inferNetwork.sh
args: ['/home/kali/Desktop/firmware-analysis-plus/firmadyne/scripts/inferNetwork.sh', '1', 'mipsel', '90']
buffer (last 100 chars): b''
before (last 100 chars): b"are 1: terminating after 90 secs...\r\nqemu-system-mipsel: -net nic,vlan=0: Invalid parameter 'vlan'\r\n"
after: <class 'pexpect.exceptions.EOF'>
match: None
match_index: None
exitstatus: None
flag_eof: True
pid: 2927189
child_fd: 5
closed: False
timeout: 30
delimiter: <class 'pexpect.exceptions.EOF'>
logfile: None
logfile_read: None
logfile_send: None
maxread: 2000
ignorecase: False
searchwindowsize: None
delaybeforesend: 0.05
delayafterclose: 0.1
delayafterterminate: 0.1
searcher: searcher_string:
    0: b'Interfaces:'

```

kal1x, 2022-06-05:
您好，我在模拟TOTOLink时出现了上述报错。对zip压缩包和web文件都进行了仿真，但是均失败了。请您拨冗查看一下错误原因，万分感谢。

liyansong2018, 2022-06-06:
根据日志，这个固件fap不能直接支持，需要定制。定制方法可参考主页上的文章。

closed 2022-07-21 by liyansong2018