secnotes/firmware-analysis-plus #37: There was a problem installing firmware-analysis-Plus

chang9807, 2022-07-17:
您好我使用Kali2022.01进行安装 我确保安装的步骤没有问题 但是
当我用binwalk "./fap.py -q ./2.5.0/ ./testcases/wnap320_V3.7.11.4_firmware.tar " 进行编译，错误结果为 ：
[+] Firmware: wnap320_V3.7.11.4_firmware.tar
[+] Extracting the firmware...
[!] Image extraction failed       

当我使用重新打包的方法 错误记录为：
+] Firmware: test.tar.gz
[+] Extracting the firmware...
[+] Cleaning previous images and created files by firmadyne
[+] All done. Go ahead and run fap.py to continue firmware analysis
[+] Image ID: 1
[+] Identifying architecture...
Traceback (most recent call last):
  File "/home/kali/桌面/firmware-analysis-plus/./fap.py", line 185, in <module>
    main()
  File "/home/kali/桌面/firmware-analysis-plus/./fap.py", line 178, in main
    arch = identify_arch(image_id)
  File "/home/kali/桌面/firmware-analysis-plus/./fap.py", line 95, in identify_arch
    child.expect_exact(":")
  File "/usr/lib/python3/dist-packages/pexpect/spawnbase.py", line 421, in expect_exact
    return exp.expect_loop(timeout)
  File "/usr/lib/python3/dist-packages/pexpect/expect.py", line 179, in expect_loop
    return self.eof(e)
  File "/usr/lib/python3/dist-packages/pexpect/expect.py", line 122, in eof
    raise exc
pexpect.exceptions.EOF: End Of File (EOF). Exception style platform.
<pexpect.pty_spawn.spawn object at 0x7f2fbea4dd00>
command: /home/kali/桌面/firmware-analysis-plus/firmadyne/scripts/getArch.sh
args: ['/home/kali/桌面/firmware-analysis-plus/firmadyne/scripts/getArch.sh', '/home/kali/桌面/firmware-analysis-plus/firmadyne/images/1.tar.gz']
buffer (last 100 chars): b''
before (last 100 chars): ''
after: <class 'pexpect.exceptions.EOF'>
match: None
match_index: None
exitstatus: None
flag_eof: True
pid: 45480
child_fd: 5
closed: False
timeout: 30
delimiter: <class 'pexpect.exceptions.EOF'>
logfile: None
logfile_read: None
logfile_send: None
maxread: 2000
ignorecase: False
searchwindowsize: None
delaybeforesend: 0.05
delayafterclose: 0.1
delayafterterminate: 0.1
searcher: searcher_string:
    0: b':'

我不太清楚是哪里有问题，您可以帮我看一下吗？

liyansong2018, 2022-07-18:
初步判断，Kali 下自带的 binwalk 好像又失效了，很多固件无法提取出文件系统。

liyansong2018, 2022-07-18:
kali 2022 自带的 binwalk 貌似还是有点问题，好像很多固件无法提取出文件系统。

liyansong2018, 2022-07-18:
鉴于Kali上的诸多bug，是时候考虑移植到Ubuntu以及docker了。

chang9807, 2022-07-19:
感谢您！

closed 2022-07-19 by chang9807

liyansong2018, 2022-07-19:
@chang9807 项目已更新，已经提供了一个临时的解决方案
**Ubuntu16.04下，使用Fap的无binwalk模式。** 后续会陆续支持binwalk的API。