secnotes/firmware-analysis-plus #43: 仿真曾经成功过的固件出现问题

chang9807, 2022-07-22:
你好！我仿真曾经成功过的固件iot_dir880l_110b01.bin（项目中自带）出现了问题 请你帮我看一下具体是哪里故障了。
错误报告如下，谢谢！

[!] Directory /qemu-builds/2.5.0 not found
[+] Using system qemu
[+] Firmware: iot_dir880l_110b01.bin
[+] Extracting the firmware...
[+] Image ID: 2
[+] Identifying architecture...
[+] Architecture: armel
[+] Building QEMU disk image...
[+] Setting up the network connection, please standby...
/home/kali/Desktop/firmware-analysis-plus/firmadyne/scripts/inferNetwork.sh
---------
['2', 'armel', '60']
---------
/home/kali/Desktop/firmware-analysis-plus/firmadyne
Traceback (most recent call last):
  File "/home/kali/Desktop/firmware-analysis-plus/./fap.py", line 185, in <module>
    main()
  File "/home/kali/Desktop/firmware-analysis-plus/./fap.py", line 180, in main
    infer_network(arch, image_id, args.time, qemu_dir)
  File "/home/kali/Desktop/firmware-analysis-plus/./fap.py", line 124, in infer_network
    child.expect_exact("Interfaces:", timeout=None)
  File "/usr/lib/python3/dist-packages/pexpect/spawnbase.py", line 421, in expect_exact
    return exp.expect_loop(timeout)
  File "/usr/lib/python3/dist-packages/pexpect/expect.py", line 179, in expect_loop
    return self.eof(e)
  File "/usr/lib/python3/dist-packages/pexpect/expect.py", line 122, in eof
    raise exc
pexpect.exceptions.EOF: End Of File (EOF). Exception style platform.
<pexpect.pty_spawn.spawn object at 0x7fbb17391eb0>
command: /home/kali/Desktop/firmware-analysis-plus/firmadyne/scripts/inferNetwork.sh
args: ['/home/kali/Desktop/firmware-analysis-plus/firmadyne/scripts/inferNetwork.sh', '2', 'armel', '60']
buffer (last 100 chars): b''
before (last 100 chars): b'irmware-analysis-plus/firmadyne/scripts//run.armel.sh: line 25: qemu-system-arm: command not found\r\n'
after: <class 'pexpect.exceptions.EOF'>
match: None
match_index: None
exitstatus: None
flag_eof: True
pid: 13966
child_fd: 5
closed: False
timeout: 30
delimiter: <class 'pexpect.exceptions.EOF'>
logfile: None
logfile_read: None
logfile_send: None
maxread: 2000
ignorecase: False
searchwindowsize: None
delaybeforesend: 0.05
delayafterclose: 0.1
delayafterterminate: 0.1
searcher: searcher_string:
    0: b'Interfaces:'

chang9807, 2022-07-22:
另外使用docker 显示没有这个库
└─$ sudo docker push liyansong2022/fap-docker:2.3.1
The push refers to repository [docker.io/liyansong2022/fap-docker]
An image does not exist locally with the tag: liyansong2022/fap-docker

chang9807, 2022-07-22:
我更新了一下项目 重新运行后错误报告为：
[!] Directory /qemu-builds/2.5.0 not found
[+] Using system qemu
[+] Firmware: iot_dir880l_110b01.bin
[+] Extracting the firmware...
[+] Image ID: 1
[+] Identifying architecture...
[+] Architecture: armel
[+] Building QEMU disk image...
[+] Setting up the network connection, please standby...
Traceback (most recent call last):
  File "/home/kali/Desktop/firmware-analysis-plus/./fap.py", line 210, in <module>
    main()
  File "/home/kali/Desktop/firmware-analysis-plus/./fap.py", line 205, in main
    infer_network(arch, image_id, args.time, qemu_dir, host)
  File "/home/kali/Desktop/firmware-analysis-plus/./fap.py", line 136, in infer_network
    child.expect_exact("Interfaces:", timeout=None)
  File "/usr/lib/python3/dist-packages/pexpect/spawnbase.py", line 421, in expect_exact
    return exp.expect_loop(timeout)
  File "/usr/lib/python3/dist-packages/pexpect/expect.py", line 179, in expect_loop
    return self.eof(e)
  File "/usr/lib/python3/dist-packages/pexpect/expect.py", line 122, in eof
    raise exc
pexpect.exceptions.EOF: End Of File (EOF). Exception style platform.
<pexpect.pty_spawn.spawn object at 0x7face0bcbf40>
command: /home/kali/Desktop/firmware-analysis-plus/firmadyne/scripts/inferNetwork.sh
args: ['/home/kali/Desktop/firmware-analysis-plus/firmadyne/scripts/inferNetwork.sh', '1', 'armel', '60']
buffer (last 100 chars): b''
before (last 100 chars): b'irmware-analysis-plus/firmadyne/scripts//run.armel.sh: line 25: qemu-system-arm: command not found\r\n'
after: <class 'pexpect.exceptions.EOF'>
match: None
match_index: None
exitstatus: 127
flag_eof: True
pid: 4562
child_fd: 5
closed: False
timeout: 30
delimiter: <class 'pexpect.exceptions.EOF'>
logfile: None
logfile_read: None
logfile_send: None
maxread: 2000
ignorecase: False
searchwindowsize: None
delaybeforesend: 0.05
delayafterclose: 0.1
delayafterterminate: 0.1
searcher: searcher_string:
    0: b'Interfaces:'

liyansong2018, 2022-07-22:
> 另外使用docker 显示没有这个库 └─$ sudo docker push liyansong2022/fap-docker:2.3.1 The push refers to repository [docker.io/liyansong2022/fap-docker] An image does not exist locally with the tag: liyansong2022/fap-docker

🤣抱歉，主页 docker 命令写错了，应该是 `pull` 而不是 `push`，我等会更正一下，谢谢提醒！

liyansong2018, 2022-07-22:
> 你好！我仿真曾经成功过的固件iot_dir880l_110b01.bin（项目中自带）出现了问题 请你帮我看一下具体是哪里故障了。 错误报告如下，谢谢！
> 
> ## [!] Directory /qemu-builds/2.5.0 not found
> [+] Using system qemu
> [+] Firmware: iot_dir880l_110b01.bin
> [+] Extracting the firmware...
> [+] Image ID: 2
> [+] Identifying architecture...
> [+] Architecture: armel
> [+] Building QEMU disk image...
> [+] Setting up the network connection, please standby...
> /home/kali/Desktop/firmware-analysis-plus/firmadyne/scripts/inferNetwork.sh
> ## ['2', 'armel', '60']
> /home/kali/Desktop/firmware-analysis-plus/firmadyne Traceback (most recent call last): File "/home/kali/Desktop/firmware-analysis-plus/./fap.py", line 185, in main() File "/home/kali/Desktop/firmware-analysis-plus/./fap.py", line 180, in main infer_network(arch, image_id, args.time, qemu_dir) File "/home/kali/Desktop/firmware-analysis-plus/./fap.py", line 124, in infer_network child.expect_exact("Interfaces:", timeout=None) File "/usr/lib/python3/dist-packages/pexpect/spawnbase.py", line 421, in expect_exact return exp.expect_loop(timeout) File "/usr/lib/python3/dist-packages/pexpect/expect.py", line 179, in expect_loop return self.eof(e) File "/usr/lib/python3/dist-packages/pexpect/expect.py", line 122, in eof raise exc pexpect.exceptions.EOF: End Of File (EOF). Exception style platform. <pexpect.pty_spawn.spawn object at 0x7fbb17391eb0> command: /home/kali/Desktop/firmware-analysis-plus/firmadyne/scripts/inferNetwork.sh args: ['/home/kali/Desktop/firmware-analysis-plus/firmadyne/scripts/inferNetwork.sh', '2', 'armel', '60'] buffer (last 100 chars): b'' before (last 100 chars): b'irmware-analysis-plus/firmadyne/scripts//run.armel.sh: line 25: qemu-system-arm: command not found\r\n' after: <class 'pexpect.exceptions.EOF'> match: None match_index: None exitstatus: None flag_eof: True pid: 13966 child_fd: 5 closed: False timeout: 30 delimiter: <class 'pexpect.exceptions.EOF'> logfile: None logfile_read: None logfile_send: None maxread: 2000 ignorecase: False searchwindowsize: None delaybeforesend: 0.05 delayafterclose: 0.1 delayafterterminate: 0.1 searcher: searcher_string: 0: b'Interfaces:'

这个是没有找到 `qemu-system-arm` ，请检查下 `-q ./qemu-builds/2.5.0` 是否拼写正确，是否在这个相对目录下有此文件。

liyansong2018, 2022-07-22:
如果没有使用 `-q` 参数，Fap 默认会使用系统存在的 `qemu-system-*` 命令模拟目标固件。
- 高版本的 Linux 发行版 apt 安装的 qemu（也是高版本） 启动系统的一些参数发生了变化，所以不建议在这些发行版上直接使用 qemu。
- Docker 环境借助于 Ubuntu 16.04，其仓库所带的 qemu 也是 2.x 版本，与 Fap 所带的大版本号一致，因此不需要添加 `-q` 指定 qemu 路径。

chang9807, 2022-07-22:
> 

太感谢啦！发现是路径找了一个点
另外我已经仿真成功 iot_dir880l_110b01.bin 
log为：
+] Firmware: iot_dir880l_110b01.bin
[+] Extracting the firmware...
[+] Image ID: 1
[+] Identifying architecture...
[+] Architecture: armel
[+] Building QEMU disk image...
[+] Setting up the network connection, please standby...
[+] Network interfaces: [('br0', '192.168.0.1'), ('br1', '192.168.7.1')]
[+] Using qemu-system-arm from /home/kali/Desktop/firmware-analysis-plus/qemu-builds/2.5.0
[+] All set! Press ENTER to run the firmware...
[+] When running, press Ctrl + A X to terminate qemu

我看到他开放了两个端口 192.168.0.1和192.168.7.1 但是我在虚拟机的浏览器输入这个网址 显示Unable to connect
请问我怎么才看到前端的显示呀。我是新手，研究生刚接触这个，还不太了解。谢谢你！！！

liyansong2018, 2022-07-22:
稍等一会就能访问了吧。注意大部分路由器的固件都是 http，而不是 https。

chang9807, 2022-07-22:
> 注意这句话
> 
> ```
> All set! Press ENTER to run the firmware...
> ```
！！！感谢！我一直忽视了这句话 怪不得！

chang9807, 2022-07-22:
您好，我想尝试仿真 firmadyne 项目已经成功仿真的固件：
但是当我仿真DIR-300A1_FW105b09.bin  wnap320_V3.7.11.4_firmware.tar 时，均显示Image提取失败，仅成功了iot_dir880l_110b01.bin一个， 请问是哪里出现了问题嘛

└─$ sudo ./fap.py -q ./qemu-builds/2.5.0/ ./testcases/DIR-300A1_FW105b09.bin

             
            ______   _                ___                 
            |  ___| (_)              / _ \                
            | |_     _   _ __ ___   / /_\ \  _ __    ___  
            |  _|   | | | '_ ` _ \  |  _  | | '_ \  / __| ++
            | |     | | | | | | | | | | | | | | | | \__ \ 
            \_|     |_| |_| |_| |_| \_| |_/ |_| |_| |___/

            Welcome to the Firmware Analysis Plus - v2.3.1
 By lys - https://github.com/liyansong2018/firmware-analysis-plus
    
[+] Firmware: DIR-300A1_FW105b09.bin
[+] Extracting the firmware...
[!] Image extraction failed

liyansong2018, 2022-07-22:
不知道你用的是哪个 Linux 发行版。#41，最新的发行版上 binwalk 存在问题，无法提取某些固件。

chang9807, 2022-07-27:
另外在仿真Firmware_TEW-411BRPplus_2.07_EU.zip 的时候出现了 Network interfaces: [] 的情况 。我在kali上也出现了同样的错误 希望老师也给看看是哪里出现的问题

[+] Firmware: WRT54G3G_2.11.05_ETSI_code.bin
[+] Extracting the firmware...
[+] Image ID: 6
[+] Identifying architecture...
[+] Architecture: mipsel
[+] Building QEMU disk image...
[+] Setting up the network connection, please standby...
[+] Network interfaces: []
[+] Using qemu-system-mipsel from /home/kali/Desktop/firmware-analysis-plus/qemu-builds/2.5.0
[+] All set! Press ENTER to run the firmware...
[+] When running, press Ctrl + A X to terminate qemu
[+] Command line: /home/kali/Desktop/firmware-analysis-plus/firmadyne/scratch/6/run.sh
Starting firmware emulation... use Ctrl-a + x to exit
kali
[    0.000000] Linux version 2.6.39.4+ (ddcc@ddcc-virtual) (gcc version 5.3.0 (GCC) ) #2 Tue Sep 1 18:11:28 EDT 2020
[    0.000000] bootconsole [early0] enabled
[    0.000000] CPU revision is: 00019300 (MIPS 24Kc)
[    0.000000] FPU revision is: 00739300
[    0.000000] Determined physical RAM map:

liyansong2018, 2022-07-28:
尝试重启一下宿主机。不同的固件Linux内核日志千差万别，仅从这些信息无法推断出具体原因。你可以看看 ./firmadyne/scratch/1/xxx.log，上面是完整的固件仿真产生的内核日志。

chang9807, 2022-07-28:
> 

老师我想问一下 除了[+] Network interfaces: []这里 还有哪里可以看网络接口吗  有很多固件模拟出来Network interfaces: 这里都是空的 换成kali也不太行

liyansong2018, 2022-07-29:
Fap 支持的固件有限，已经验证过的固件都已在主页中体现。其他固件不保证仿真的完整性。你可以通过 `./firmadyne/scratch/1/qemu.initial.serial.log` 查看完整日志，确定目标固件是否正常启动了。如果没有，则 Network interfaces 自然是推测不出来的。

closed 2022-08-30 by chang9807

Ahaobi, 2024-04-22:
mount: mounting /dev/mtdblock/4 on /htdocs/web/js/localization failed: Invalid argument
[/etc/init0.d/S40event.sh]: start ...
/dev/nvram: No such device
/dev/nvram: No such device

你好请问 我在执行命令  ./fap.py -q ./qemu-builds/2.5.0/ ./testcases/iot_dir880l_110b01.bin 报了如上错误应该如何处理呀