secnotes/firmware-analysis-plus #46: 无法执行init，导致Kernel panic

ioxv, 2022-10-11:
您好，当我对一个基于Openwrt+Luci的第三方固件进行仿真的时候，提示`Starting init: /sbin/init exists but couldn't execute it (error -8)`，进而导致Kernel panic。目前在Ubuntu20.04以及16.04环境下进行过测试，均有这个问题。请问，您有什么好的解决方案吗？

```
[+] Extracting the firmware...
[+] Image ID: 1
[+] Identifying architecture...
[+] Architecture: armel
[+] Building QEMU disk image...
[+] Setting up the network connection, please standby...
[+] [info] Running firmware 1: terminating after 60 secs...
qemu-system-arm: terminating on signal 2 from pid 12402
[+] [info] Inferring network...
[+] [info] Interfaces: []
[+] [info] Done!
[+] Using qemu-system-arm from /home/a/src/firmware-analysis-plus/qemu-builds/2.5.0
[+] All set! Press ENTER to run the firmware...
[+] When running, press Ctrl + A X to terminate qemu
[+] Command line: /home/a/src/firmware-analysis-plus/firmadyne/scratch/1/run.sh
Starting firmware emulation... use Ctrl-a + x to exit
root
Warning: hub port hub3port0 has no peer
Warning: vlan 3 with no nics
Warning: hub port hub2port0 has no peer
Warning: vlan 2 with no nics
Warning: hub port hub1port0 has no peer
Warning: vlan 1 with no nics
Warning: hub port hub0port0 has no peer
Warning: vlan 0 with no nics
Warning: netdev hub0port0 has no peer
Warning: netdev hub1port0 has no peer
Warning: netdev hub2port0 has no peer
Warning: netdev hub3port0 has no peer
Warning: requested NIC (anonymous, model unspecified) was not created (not supported by this machine?)
Warning: requested NIC (anonymous, model unspecified) was not created (not supported by this machine?)
Warning: requested NIC (anonymous, model unspecified) was not created (not supported by this machine?)
Warning: requested NIC (anonymous, model unspecified) was not created (not supported by this machine?)
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.1.17+ (vagrant@vagrant-ubuntu-trusty-64) (gcc version 5.3.0 (GCC) ) #1 Thu Feb 18 01:05:21 UTC 2016
[    0.000000] CPU: ARMv7 Processor [412fc0f1] revision 1 (ARMv7), cr=10c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
[    0.000000] Machine model: linux,dummy-virt
[    0.000000] debug: ignoring loglevel setting.
[    0.000000] Memory policy: Data cache writeback
[    0.000000] On node 0 totalpages: 65536
[    0.000000] free_area_init_node: node 0, pgdat c061dfe8, node_mem_map cfdf9000
[    0.000000]   Normal zone: 512 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 65536 pages, LIFO batch:15
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv0.2 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] CPU: All CPU(s) started in SVC mode.
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 65024
[    0.000000] Kernel command line: root=/dev/vda1 console=ttyS0 nandsim.parts=64,64,64,64,64,64,64,64,64,64 rdinit=/firmadyne/preInit.sh rw debug ignore_loglevel print-fatal-signals=1 user_debug=31 firmadyne.syscall=0
[    0.000000] PID hash table entries: 1024 (order: 0, 4096 bytes)
[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes)
[    0.000000] Memory: 253340K/262144K available (4297K kernel code, 170K rwdata, 1584K rodata, 180K init, 148K bss, 8804K reserved, 0K cma-reserved)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
[    0.000000]     vmalloc : 0xd0800000 - 0xff000000   ( 744 MB)
[    0.000000]     lowmem  : 0xc0000000 - 0xd0000000   ( 256 MB)
[    0.000000]     modules : 0xbf000000 - 0xc0000000   (  16 MB)
[    0.000000]       .text : 0xc0008000 - 0xc05c67bc   (5882 kB)
[    0.000000]       .init : 0xc05c7000 - 0xc05f4000   ( 180 kB)
[    0.000000]       .data : 0xc05f4000 - 0xc061e840   ( 171 kB)
[    0.000000]        .bss : 0xc0621000 - 0xc06462d4   ( 149 kB)
[... 181 lines cut ...]
[    0.401947] rtc-pl031 9010000.pl031: setting system clock to 2022-10-11 03:27:28 UTC (1665458848)
[    0.408850] EXT4-fs (vda1): couldn't mount as ext3 due to feature incompatibilities
[    0.411175] EXT4-fs (vda1): mounting ext2 file system using the ext4 subsystem
[    0.420496] EXT4-fs (vda1): warning: mounting unchecked fs, running e2fsck is recommended
[    0.422479] EXT4-fs (vda1): mounted filesystem without journal. Opts: (null)
[    0.423112] VFS: Mounted root (ext2 filesystem) on device 254:1.
[    0.428000] Freeing unused kernel memory: 180K (c05c7000 - c05f4000)
[    0.453004] request_module: runaway loop modprobe binfmt-464c
[    0.462294] Starting init: /sbin/init exists but couldn't execute it (error -8)
[    0.482910] request_module: runaway loop modprobe binfmt-464c
[    0.489695] Starting init: /bin/sh exists but couldn't execute it (error -8)
[    0.489875] Kernel panic - not syncing: No working init found.  Try passing init= option to kernel. See Linux Documentation/init.txt for guidance.
[    0.490119] CPU: 0 PID: 1 Comm: swapper Tainted: G        W       4.1.17+ #1
[    0.490224] Hardware name: Generic DT based system
[    0.490335] [<c001c99c>] (unwind_backtrace) from [<c0019d30>] (show_stack+0x10/0x14)
[    0.490522] [<c0019d30>] (show_stack) from [<c0410b78>] (panic+0x70/0x1c8)
[    0.490672] [<c0410b78>] (panic) from [<c040f33c>] (kernel_init+0xb8/0xe4)
[    0.490829] [<c040f33c>] (kernel_init) from [<c0016da8>] (ret_from_fork+0x14/0x2c)
[    0.491250] ---[ end Kernel panic - not syncing: No working init found.  Try passing init= option to kernel. See Linux Documentation/init.txt for guidance.
```
执行的命令如下，第二条是`ps`显示的。
```
./fap.py -q ./qemu-builds/2.5.0/ /home/a/ER3200G3-MNW210-R0118.img
/home/a/src/firmware-analysis-plus/qemu-builds/2.5.0/qemu-system-arm -m 256 -M virt -kernel /home/a/src/firmware-analysis-plus/firmadyne/binaries//zImage.armel -drive if=none,file=/home/a/src/firmware-analysis-plus/firmadyne/scratch//1//image.raw,format=raw,id=rootfs -device virtio-blk-device,drive=rootfs -append root=/dev/vda1 console=ttyS0 nandsim.parts=64,64,64,64,64,64,64,64,64,64 rdinit=/firmadyne/preInit.sh rw debug ignore_loglevel print-fatal-signals=1 user_debug=31 firmadyne.syscall=0 -nographic -net nic,vlan=0 -net socket,vlan=0,listen=:2000 -net nic,vlan=1 -net socket,vlan=1,listen=:2001 -net nic,vlan=2 -net socket,vlan=2,listen=:2002 -net nic,vlan=3 -net socket,vlan=3,listen=:2003
```

liyansong2018, 2022-10-11:
抱歉，由于当前大多数IoT固件为32位，因此Fap只支持arm32和mips32架构，不支持64bit。

ioxv, 2022-10-12:
好吧，期待相关更新。（不过换个内核镜像，手动qemu能运行，但是ubus存在问题）

closed 2022-10-12 by ioxv