secnotes/firmware-analysis-plus #47: 仿真后网络接口为空

zhaollj, 2022-12-15:
请问固件仿真成功后网络接口为空是因为什么呀？还有就是如果存在网络接口的话，每次要以什么方式打开呢，自己在浏览器中输入吗？
[image]

liyansong2018, 2022-12-15:
DIR2640需要用定制版 [Fap 2.0.1](https://github.com/liyansong2018/firmware-analysis-plus/releases/tag/v2.0.1)，其他固件可以用最新版。

zhaollj, 2022-12-15:
那请问是实现自动打开管理页面呢？还是需要每次手动输入呀？我这边好像每次都得自己手动输入

zhaollj, 2022-12-15:
还有就是我使用定制版Fap2.0.1对DIR2640进行仿真的时候，出现了错误，使用的系统是ubuntu20.04,请问是什么原因呢？
[image]
[image]
[image]

Isolator-1, 2022-12-16:
请问我使用的是库里testcases路径下的固件，也没有network interfaces，我也试了一下别的固件，这个列表依然是空的，这可能是什么原因导致的呀？谢谢
[image]

liyansong2018, 2022-12-16:
首先看看binwalk能不能解压这个固件吧，理论上不应该出现这种情况。最新的 binwalk 貌似某些固件不能解压。

liyansong2018, 2022-12-16:
@Isolator-1 你的问题是由于使用 `fap.py` 没有加参数，只有在Ubuntu16.04系统下，可以省略 `-q ./qemu-builds/2.5.0/`

liyansong2018, 2022-12-16:
@zhaollj 抱歉，你的问题是由于专门针对DIR2640路由器的定制版Fap2.0.1缺少某些补丁，后续没有维护了。当时这个版本旨在Kali2020上测试通过。

hy-2333, 2023-04-07:
[image]
我这边也是没有网络接口的地址 甚至还是用的同样版本的wnap320

liyansong2018, 2023-04-07:
@hy-2333 
- 你用的操作系统 Ubuntu 哪个版本的
- `firmadyne/scratch/num/` 路径下有详细日志

hy-2333, 2023-04-07:
> 

18.04

liyansong2018, 2023-04-07:
可以粘贴一下详细日志

hy-2333, 2023-04-07:
> 

我的日志打开后是空的 显示照片源文件丢失...
然后我刚刚重新跑了一遍 又出现了新的问题...什么都没做...

hy-2333, 2023-04-07:
[image]
[image]
感觉莫名其妙的

liyansong2018, 2023-04-07:
看起来似乎是因为有端口被占用了，运行 `./shutdown.py`，或者重启操作系统，也许能解决上面的问题。至于你最初提的仿真接口为空，这只是表面现象，说明仿真失败了，需要详细日志支撑。Fap 前两年已迭代多个版本，理论上不应该再出现你说的这些问题。

hy-2333, 2023-04-07:
[image]
我是从博主的firmware-analysis-toolkit那篇文章过来的 在那篇文章中也是卡在最后这个问题 照您说的重定向路径后仍然报相同错误

hy-2333, 2023-04-07:
> 看起来似乎是因为有端口被占用了，运行 `./shutdown.py`，或者重启操作系统，也许能解决上面的问题。至于你最初提的仿真接口为空，这只是表面现象，说明仿真失败了，需要详细日志支撑。Fap 前两年已迭代多个版本，理论上不应该再出现你说的这些问题。

[qemu.initial.serial.log](https://github.com/liyansong2018/firmware-analysis-plus/files/11178685/qemu.initial.serial.log)
好吧 端口关掉后仍然是网络接口为空 这次导出日志了

liyansong2018, 2023-04-07:
抱歉哭瞎🤣从日志上看，固件已经启动了，没有发现明显问题。
- 重启一下你的Ubuntu，可能是你系统网卡的问题，不知道你之前有没有配置过网络，可能更改了Linux原始的网卡配置
- 更换操作系统或者使用Docker，再尝试下

hy-2333, 2023-04-07:
> 

博主你好 我这边在新开的Ubuntu系统下试了一下也是网口为空... 我这边是用vmware开得虚拟机 不知道这是不是有什么影响

liyansong2018, 2023-04-07:
> > 
> 
> 博主你好 我这边在新开的Ubuntu系统下试了一下也是网口为空... 我这边是用vmware开得虚拟机 不知道这是不是有什么影响

不会有影响，我等会 Ubuntu18.04 的环境。

liyansong2018, 2023-04-07:
@hy-2333 
- **请再次确认你的安装步骤有没有问题**，是否按照主页上的步骤来进行测试的。
- 我刚刚在Vmware上安装了Ubuntu18.04，没有发现问题，并且**此问题早已修复**，修复之后也没有其他同学反馈同样的问题。
```shell
tom@ubuntu18:~/Documents/firmware-analysis-plus-2.3.1$ ./fap.py -q ./qemu-builds/2.5.0/ ./testcases/iot_dir880l_110b01.bin 

             
            ______   _                ___                 
            |  ___| (_)              / _ \                
            | |_     _   _ __ ___   / /_\ \  _ __    ___  
            |  _|   | | | '_ ` _ \  |  _  | | '_ \  / __| ++
            | |     | | | | | | | | | | | | | | | | \__ \ 
            \_|     |_| |_| |_| |_| \_| |_/ |_| |_| |___/

            Welcome to the Firmware Analysis Plus - v2.3.1
 By lys - https://github.com/liyansong2018/firmware-analysis-plus
    
[+] Firmware: iot_dir880l_110b01.bin
[+] Extracting the firmware...
[+] Image ID: 1
[+] Identifying architecture...
[+] Architecture: armel
[+] Building QEMU disk image...
[+] Setting up the network connection, please standby...
[+] Network interfaces: [('br0', '192.168.0.1'), ('br1', '192.168.7.1')]
[+] Using qemu-system-arm from /home/tom/Documents/firmware-analysis-plus-2.3.1/qemu-builds/2.5.0
[+] All set! Press ENTER to run the firmware...
[+] When running, press Ctrl + A X to terminate qemu
```

hy-2333, 2023-04-07:
> 

好吧 感谢博主 我这边已经解决了

liyansong2018, 2023-04-07:
@hy-2333

不客气，所以你是怎么解决的

hy-2333, 2023-04-07:
> @hy-2333
> 
> 不客气，所以你是怎么解决的

我这边新的Ubuntu刚开始配了那些源后网掉了...

hy-2333, 2023-04-07:
> @hy-2333
> 
> 不客气，所以你是怎么解决的

把binwalk，firmadyne那些全用的firmaware plus里自带的就和usage里一样了

liyansong2018, 2023-04-07:
😀解决就好，果然是你 Ubuntu 系统网络问题，导致某些依赖没有安装好。

closed 2023-04-07 by liyansong2018

sdahiasdba, 2024-04-02:
__           _
                              / _|         | |
                             | |_    __ _  | |_
                             |  _|  / _` | | __|
                             | |   | (_| | | |_
                             |_|    \__,_|  \__|

                Welcome to the Firmware Analysis Toolkit - v0.3
    Offensive IoT Exploitation Training http://bit.do/offensiveiotexploitation
                  By Attify - https://attify.com  | @attifyme
    
[+] Firmware: Gaw5.6T02-4-DL-R1B010-ME.EN_whole_image(0222163812) (1).en_whole_image
[+] Extracting the firmware...
[+] Image ID: 1
[+] Identifying architecture...
[+] Architecture: mipseb
[+] Building QEMU disk image...
[+] Setting up the network connection, please standby...
[+] Network interfaces: []
[+] All set! Press ENTER to run the firmware...
[+] When running, press Ctrl + A X to terminate qemu

@liyansong2018 anyone solve this problem

Hwe1916, 2025-04-26:
> > [@hy-2333](https://github.com/hy-2333)
> > 不客气，所以你是怎么解决的
> 
> 把binwalk，firmadyne那些全用的firmaware plus里自带的就和usage里一样了

可以具体讲一下怎么操作的吗