secnotes/firmware-analysis-plus #48: [!] Cannot emulate firmware, run.sh not generated

000xE4, 2023-02-03:
请问这个问题是什么原因导致的呢
|>23:48:09<|(^ω^)$python3 fap.py  -q  qemu-builds/2.5.0/ US_A15V1.0RTL_V15.13.07.13_multi_TD01.bin

            ______   _                ___
            |  ___| (_)              / _ \
            | |_     _   _ __ ___   / /_\ \  _ __    ___
            |  _|   | | | '_ ` _ \  |  _  | | '_ \  / __| ++
            | |     | | | | | | | | | | | | | | | | \__ \
            \_|     |_| |_| |_| |_| \_| |_/ |_| |_| |___/

            Welcome to the Firmware Analysis Plus - v2.3
 By lys - https://github.com/liyansong2018/firmware-analysis-plus

[+] Firmware: US_A15V1.0RTL_V15.13.07.13_multi_TD01.bin
[+] Extracting the firmware...
[+] Image ID: 1
[+] Identifying architecture...
[+] Architecture: mipsel
[+] Building QEMU disk image...
[+] Setting up the network connection, please standby...
[+] [info] Running firmware 1: terminating after 60 secs...
qemu-system-mipsel: -drive if=ide,format=raw,file=/mnt/hgfs/downtest/firmware-analysis-plus-2.3/firmadyne/scratch//1//image.raw: Could not open '/mnt/hgfs/downtest/firmware-analysis-plus-2.3/firmadyne/scratch//1//image.raw': No such file or directory
[!] Cannot emulate firmware, run.sh not generated

liyansong2018, 2023-02-03:
能提供一下这个固件么，我需要看一下详细的日志。

000xE4, 2023-02-03:
您好，我已经收到了您的邮件。

000xE4, 2023-02-03:
链接: https://pan.baidu.com/s/1qqpuW8ECRWLE765p-K__3g 提取码: wflh
麻烦师傅了，我用了项目中自带的测试固件也存在这个问题。

liyansong2018, 2023-02-04:
我在Kali如下发行版中，暂未重现你的问题，不知道你用的是什么发行版
```
$ lsb_release -a
No LSB modules are available.
Distributor ID: Kali
Description:    Kali GNU/Linux Rolling
Release:        2022.1
Codename:       kali-rolling
```

建议

1.用最新版的`Fap`，已经更新到2.3.1，之前忘了发布release包，我现在重新打包一下
```shell
# 有git环境
git clone https://github.com/liyansong2018/firmware-analysis-plus.git
# 无git环境
https://github.com/liyansong2018/firmware-analysis-plus/releases/tag/v2.3.1
```
2.如上述问题未解决，尝试更换发行版为项目主页列表上所支持的，此类Linux已经经过测试

3.如还有问题，可以尝试使用docker，主页也给出了docker的使用方式

4.如上述问题还未解决，发送一下详细日志，在此路径 `scratch/1/xxx.log`，我可以定位一下问题根因。

000xE4, 2023-02-04:
我用的是ubuntu16，下载的是最新版的fap 2.3，1，我尝试一下kali版本吧，谢谢师傅！！！
No LSB modules are available.
Distributor ID:    Ubuntu
Description:    Ubuntu 16.04.4 LTS
Release:    16.04
Codename:    xenial