secnotes/firmware-analysis-plus #54: 请教项目运行的2个错误（pexpect.exceptions.EOF、访问不到8080）

abcdefg-png, 2023-12-24:
我使用的是Ubuntu18.04，python3.8，postgresql数据库密码为firmadyne
问题1，使用python3.8（python2.7也一样）启动fap，：
```sh
service postgresql start # 启动数据库
(fat_py2) yhr@ubuntu:~/Desktop/rooter/fat/Old-Firmware-Analysis-Toolkit/firmadyne$ python fat.py
(fat_py3) yhr@ubuntu:~/Desktop/rooter/fat/firmware-analysis-plus$ ./fap.py -q ./qemu-builds/2.5.0/ /home/yhr/Desktop/rooter/firmwares/wnap320_V3.7.11.4_firmware.tar
```
失败的报错如下：
```sh
(fat_py3) yhr@ubuntu:~/Desktop/rooter/fat/firmware-analysis-plus$ ./fap.py -q ./qemu-builds/2.5.0/ /home/yhr/Desktop/rooter/firmwares/wnap320_V3.7.11.4_firmware.tar
……
[+] Firmware: wnap320_V3.7.11.4_firmware.tar
[+] Extracting the firmware...
Traceback (most recent call last):
  File "./fap.py", line 210, in <module>
    main()
  File "./fap.py", line 198, in main
    image_id = run_extractor(args.firm_path, args.binwalk, host)
  File "./fap.py", line 58, in run_extractor
    child.expect_exact("Tag: ")
  File "/home/yhr/anaconda3/envs/fat_py3/lib/python3.8/site-packages/pexpect-4.9.0-py3.8.egg/pexpect/spawnbase.py", line 432, in expect_exact
    return exp.expect_loop(timeout)
  File "/home/yhr/anaconda3/envs/fat_py3/lib/python3.8/site-packages/pexpect-4.9.0-py3.8.egg/pexpect/expect.py", line 179, in expect_loop
    return self.eof(e)
  File "/home/yhr/anaconda3/envs/fat_py3/lib/python3.8/site-packages/pexpect-4.9.0-py3.8.egg/pexpect/expect.py", line 122, in eof
    raise exc
pexpect.exceptions.EOF: End Of File (EOF). Exception style platform.
<pexpect.pty_spawn.spawn object at 0x7f76237cc4f0>
command: /home/yhr/anaconda3/envs/fat_py3/bin/python3
args: ['/home/yhr/anaconda3/envs/fat_py3/bin/python3', '/home/yhr/Desktop/rooter/fat/firmware-analysis-plus/firmadyne/sources/extractor/extractor.py', '-np', '-nk', '/home/yhr/Desktop/rooter/firmwares/wnap320_V3.7.11.4_firmware.tar', '/home/yhr/Desktop/rooter/fat/firmware-analysis-plus/firmadyne/images']
buffer (last 100 chars): b''
before (last 100 chars): b'xtractor.py", line 17, in <module>\r\n    import magic\r\nModuleNotFoundError: No module named \'magic\'\r\n'
after: <class 'pexpect.exceptions.EOF'>
……
```
问题2，后来我换了dockers，问题1解决了，但运行起来以后访问不到127.0.0.1端口。
```sh
./fap.py testcases/iot_dir880l_110b01.bin
```
[image]
感觉fap工具很有意义，十分钦佩您们的工作，期待您的回复！

liyansong2018, 2024-03-17:
使用docker的话，需要自己映射网络端口，这样宿主机才能访问容器的端口。项目主页有提到端口映射的方法。

liyansong2018, 2024-05-29:
可以直接使用打包好的 [vmware镜像](https://pan.baidu.com/s/1eVNxoLKlqAQHcrSMfI7tQw?pwd=jpy4)。

closed 2024-05-29 by liyansong2018