secnotes/firmware-analysis-plus #58: 宿主机无妨访问

dctorwho, 2024-05-24:
[image]
[image]
[image]
这是ifconfig的结果：
[image]
这是ip route的结果：
[image]
以及我尝试桥接模式，还是ping不通

liyansong2018, 2024-05-29:
由于docker更新，当前确实存在此问题。推荐使用Ubuntu20.04，直接使用fap。

liyansong2018, 2024-05-29:
可以直接使用打包好的 [vmware镜像](https://pan.baidu.com/s/1eVNxoLKlqAQHcrSMfI7tQw?pwd=jpy4)。

closed 2024-05-29 by liyansong2018