secnotes/firmware-analysis-plus #59: 运行时报错image_id = run_extractor(args.firm_path, args.binwalk, host)

xiaodian2, 2024-06-21:
已经确定固件可以被解压成功，但是运行报错，fat也是一样的，binwalk使用的是您的版本，已经重装过很多次了，但是还是有问题，Ubuntu20.04。
kali@ubuntu:~/Desktop/firmware-analysis-plus$ python3 ./fap.py -q ./qemu-builds/2.5.0/ ./netcore_COVER5_V1.3.240102.100611.bin 

             
            ______   _                ___                 
            |  ___| (_)              / _ \                
            | |_     _   _ __ ___   / /_\ \  _ __    ___  
            |  _|   | | | '_ ` _ \  |  _  | | '_ \  / __| ++
            | |     | | | | | | | | | | | | | | | | \__ \ 
            \_|     |_| |_| |_| |_| \_| |_/ |_| |_| |___/

            Welcome to the Firmware Analysis Plus - v2.3.1
 By lys - https://github.com/liyansong2018/firmware-analysis-plus
    
[+] Firmware: netcore_COVER5_V1.3.240102.100611.bin
[+] Extracting the firmware...
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/pexpect/spawnbase.py", line 166, in read_nonblocking
    s = os.read(self.child_fd, size)
OSError: [Errno 5] Input/output error

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/pexpect/expect.py", line 111, in expect_loop
    incoming = spawn.read_nonblocking(spawn.maxread, timeout)
  File "/usr/lib/python3/dist-packages/pexpect/pty_spawn.py", line 485, in read_nonblocking
    return super(spawn, self).read_nonblocking(size)
  File "/usr/lib/python3/dist-packages/pexpect/spawnbase.py", line 171, in read_nonblocking
    raise EOF('End Of File (EOF). Exception style platform.')
pexpect.exceptions.EOF: End Of File (EOF). Exception style platform.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "./fap.py", line 210, in <module>
    main()
  File "./fap.py", line 198, in main
    image_id = run_extractor(args.firm_path, args.binwalk, host)
  File "./fap.py", line 58, in run_extractor
    child.expect_exact("Tag: ")
  File "/usr/lib/python3/dist-packages/pexpect/spawnbase.py", line 418, in expect_exact
    return exp.expect_loop(timeout)
  File "/usr/lib/python3/dist-packages/pexpect/expect.py", line 117, in expect_loop
    return self.eof(e)
  File "/usr/lib/python3/dist-packages/pexpect/expect.py", line 63, in eof
    raise EOF(msg)
pexpect.exceptions.EOF: End Of File (EOF). Exception style platform.
<pexpect.pty_spawn.spawn object at 0x7f2f91543d00>
command: /usr/bin/python3
args: ['/usr/bin/python3', '/home/kali/Desktop/firmware-analysis-plus/firmadyne/sources/extractor/extractor.py', '-np', '-nk', './netcore_COVER5_V1.3.240102.100611.bin', '/home/kali/Desktop/firmware-analysis-plus/firmadyne/images']
buffer (last 100 chars): b''
before (last 100 chars): b' your installation")\r\nImportError: python-magic: failed to find libmagic.  Check your installation\r\n'
after: <class 'pexpect.exceptions.EOF'>
match: None
match_index: None
exitstatus: None
flag_eof: True
pid: 49753
child_fd: 5
closed: False
timeout: None
delimiter: <class 'pexpect.exceptions.EOF'>
logfile: None
logfile_read: None
logfile_send: None
maxread: 2000
ignorecase: False
searchwindowsize: None
delaybeforesend: 0.05
delayafterclose: 0.1
delayafterterminate: 0.1
searcher: searcher_string:
    0: b'Tag: '

liyansong2018, 2024-06-23:
单从这里有限日志，只能看出来缺少某些依赖，可以试试已经打包好的VMware镜像。
另外，我在`Kali 2023.3`中测试了你的固件，环境没有问题，但是fap无法仿真你的固件。
```shell
$ ./fap.py -q ./qemu-builds/2.5.0/ ./testcases/netcore_COVER5_V1.3.240102.100611.bin 

             
            ______   _                ___                 
            |  ___| (_)              / _ \                
            | |_     _   _ __ ___   / /_\ \  _ __    ___  
            |  _|   | | | '_ ` _ \  |  _  | | '_ \  / __| ++
            | |     | | | | | | | | | | | | | | | | \__ \ 
            \_|     |_| |_| |_| |_| \_| |_/ |_| |_| |___/

            Welcome to the Firmware Analysis Plus - v2.3.1
 By lys - https://github.com/liyansong2018/firmware-analysis-plus
    
[+] Firmware: netcore_COVER5_V1.3.240102.100611.bin
[+] Extracting the firmware...
[+] Image ID: 2
[+] Identifying architecture...
[+] Architecture: mipsel
[+] Building QEMU disk image...
[+] Setting up the network connection, please standby...
[+] Network interfaces: []
[+] Using qemu-system-mipsel from /home/kali/firmware-analysis-plus/qemu-builds/2.5.0
[+] All set! Press ENTER to run the firmware...
[+] When running, press Ctrl + A X to terminate qemu
```
根据日志，内核无法正确挂在根文件系统
```
[    4.492000] mount_root: failed to launch kmodloader from internal overlay
[    4.528000] do_page_fault() #2: sending SIGSEGV to block for invalid read access from
```

liyansong2018, 2024-06-23:
Fap能力有限，如果是想研究目标固件的核心进程，建议直接模拟单个进程，可参考
- [固件中的单个二进制模拟：Tenda AC15 路由器 CVE-2018-5767 / CVE-2020-10987 漏洞分析与复现](https://liyansong.blog.csdn.net/article/details/113800058)
- [小米路由器固件仿真模拟方案](https://bbs.kanxue.com/thread-282034.htm)

xiaodian2, 2024-06-23:
感谢您的解答，我直接使用您提供的虚拟机了，仿真其他固件完全没有问题，再次感谢

closed 2024-06-23 by xiaodian2