secnotes/firmware-analysis-plus #71: 虚拟机镜像

ZaneBoden, 2025-04-22:
你好，我下载了提供的虚拟机镜像，但是由于用户密码未知，无法使用，望告知。
谢谢

liyansong2018, 2025-04-22:
密码是，toor。参考 #64

ZaneBoden, 2025-04-23:
感谢回复！
使用了tenda固件进行测试，没有网络接口，我看了之前issue里的解决办法，没有头绪。
希望作者大大能提供一些指导意见
[+] Firmware: NX15V100R011.bin
[+] Extracting the firmware...
[+] Image ID: 1
[+] Identifying architecture...
[+] Architecture: mipsel
[+] Building QEMU disk image...
[+] Setting up the network connection, please standby...
[+] Network interfaces: []
[+] Using qemu-system-mipsel from /home/jerry/firmware-analysis-plus/qemu-builds/2.5.0
[+] All set! Press ENTER to run the firmware...
[+] When running, press Ctrl + A X to terminate qemu

ZaneBoden, 2025-04-23:
对应的日志文件qemu.initial.serial.log内容如下：
[    0.000000] Linux version 2.6.39.4+ (ddcc@ddcc-virtual) (gcc version 5.3.0 (GCC) ) #2 Tue Sep 1 18:11:28 EDT 2020
[    0.000000] bootconsole [early0] enabled
[    0.000000] CPU revision is: 00019300 (MIPS 24Kc)
[    0.000000] FPU revision is: 00739300
[    0.000000] Determined physical RAM map:
[    0.000000]  memory: 00001000 @ 00000000 (reserved)
[    0.000000]  memory: 000ef000 @ 00001000 (ROM data)
[    0.000000]  memory: 0065e000 @ 000f0000 (reserved)
[    0.000000]  memory: 0f8b2000 @ 0074e000 (usable)
[    0.000000] debug: ignoring loglevel setting.
[    0.000000] Wasting 59840 bytes for tracking 1870 unused pages
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] Zone PFN ranges:
[    0.000000]   DMA      0x00000000 -> 0x00001000
[    0.000000]   Normal   0x00001000 -> 0x00010000
[    0.000000] Movable zone start PFN for each node
[    0.000000] early_node_map[1] active PFN ranges
[    0.000000]     0: 0x00000000 -> 0x00010000
[    0.000000] On node 0 totalpages: 65536
[    0.000000] free_area_init_node: node 0, pgdat 806e8800, node_mem_map 81000000
[    0.000000]   DMA zone: 32 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 4064 pages, LIFO batch:0
[    0.000000]   Normal zone: 480 pages used for memmap
[    0.000000]   Normal zone: 60960 pages, LIFO batch:15
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0 
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 65024
[    0.000000] Kernel command line: firmadyne.syscall=1 root=/dev/sda1 console=ttyS0 nandsim.parts=64,64,64,64,64,64,64,64,64,64 rdinit=/firmadyne/preInit.sh rw debug ignore_loglevel print-fatal-signals=1
[    0.000000] PID hash table entries: 1024 (order: 0, 4096 bytes)
[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes)
[    0.000000] Primary instruction cache 2kB, VIPT, 2-way, linesize 16 bytes.
[    0.000000] Primary data cache 2kB, 2-way, VIPT, no aliases, linesize 16 bytes
[    0.000000] Writing ErrCtl register=00000000
[    0.000000] Readback ErrCtl register=00000000
[    0.000000] Memory: 252372k/254664k available (4453k kernel code, 2292k reserved, 1606k data, 240k init, 0k highmem)
[    0.000000] NR_IRQS:256
[    0.000000] CPU frequency 799.37 MHz
[    0.000000] Console: colour dummy device 80x25
[    0.000000] Calibrating delay loop... 3307.52 BogoMIPS (lpj=6615040)
[    0.052000] pid_max: default: 32768 minimum: 301
[    0.052000] Mount-cache hash table entries: 512
[    0.056000] Performance counters: No available PMU.
[    0.056000] NET: Registered protocol family 16
[    0.064000] bio: create slab <bio-0> at 0
[    0.064000] vgaarb: loaded
[    0.064000] SCSI subsystem initialized
[    0.064000] libata version 3.00 loaded.
[    0.064000] usbcore: registered new interface driver usbfs
[    0.068000] usbcore: registered new interface driver hub
[    0.068000] usbcore: registered new device driver usb
[    0.068000] pci 0000:00:00.0: [11ab:4620] type 0 class 0x000600
[    0.068000] pci 0000:00:00.0: reg 14: [mem 0x01000000-0x01ffffff pref]
[    0.068000] pci 0000:00:0a.0: [8086:7110] type 0 class 0x000601
[    0.068000] pci 0000:00:0a.1: [8086:7111] type 0 class 0x000101
[    0.068000] pci 0000:00:0a.1: reg 20: [io  0x0000-0x000f]
[    0.068000] pci 0000:00:0a.2: [8086:7112] type 0 class 0x000c03
[    0.068000] pci 0000:00:0a.2: reg 20: [io  0x0000-0x001f]
[    0.068000] pci 0000:00:0a.3: [8086:7113] type 0 class 0x000680
[    0.068000] pci 0000:00:0a.3: address space collision: [io  0x1100-0x110f] conflicts with GT-64120 PCI I/O [io  0x1000-0x1fffff]
[    0.068000] pci 0000:00:0b.0: [1022:2000] type 0 class 0x000200
[    0.068000] pci 0000:00:0b.0: reg 10: [io  0x0000-0x001f]
[    0.072000] pci 0000:00:0b.0: reg 14: [mem 0x00000000-0x0000001f]
[    0.072000] pci 0000:00:0b.0: reg 30: [mem 0x00000000-0x0003ffff pref]
[    0.072000] pci 0000:00:12.0: [1022:2000] type 0 class 0x000200
[    0.072000] pci 0000:00:12.0: reg 10: [io  0x0000-0x001f]
[    0.072000] pci 0000:00:12.0: reg 14: [mem 0x00000000-0x0000001f]
[    0.072000] pci 0000:00:12.0: reg 30: [mem 0x00000000-0x0003ffff pref]
[... 200 lines cut ...]
[    0.740000] sd 0:0:0:0: [sda] 2097152 512-byte logical blocks: (1.07 GB/1.00 GiB)
[    0.740000] sd 0:0:0:0: [sda] Write Protect is off
[    0.744000] sd 0:0:0:0: [sda] Mode Sense: 00 3a 00 00
[    0.744000] sd 0:0:0:0: [sda] Write cache: enabled, read cache: enabled, doesn't support DPO or FUA
[    0.752000]  sda: sda1
[    0.756000] sd 0:0:0:0: [sda] Attached SCSI disk
[    0.896000] input: ImExPS/2 Generic Explorer Mouse as /devices/platform/i8042/serio1/input/input1
[    0.924000] VFS: Mounted root (ext2 filesystem) on device 8:1.
[    0.924000] Freeing prom memory: 956k freed
[    0.932000] Freeing unused kernel memory: 240k freed
[    1.024000] init: Console is alive
[    2.032000] init: - preinit -
[    2.032000] firmadyne: do_execve: /firmadyne/console
[    2.032000] OFFSETS: offset of pid: 0xc4 offset of comm: 0x1a4
[    2.056000] firmadyne: sys_socket[PID: 55 (procd)]: family:16, type:524289, protocol:15
[    2.056000] firmadyne: sys_setsockopt[PID: 55 (procd)]: fd:6, level:65535, optname:33
[    2.204000] mount_root: BUG: no suitable fs found
overlays mount failed reboot
reboot
[    4.232000] firmadyne: sys_reboot[PID: 1 (init)]: magic1:fee1dead, magic2:28121969, cmd:1234567
[    4.232000] firmadyne: sys_reboot: removed CAP_SYS_BOOT, starting init...
[    5.268000] init: - preinit -
[    5.328000] firmadyne: sys_socket[PID: 64 (procd)]: family:16, type:524289, protocol:15
[    5.344000] firmadyne: sys_setsockopt[PID: 64 (procd)]: fd:3, level:65535, optname:33
[    5.420000] mount_root: BUG: no suitable fs found

closed 2025-06-22 by ZaneBoden